libusb/libusb #303: libusb_get_port_numbers is broken on Mac OS Sierra 10.12.4 (16E195)

mvp, 2017-05-24:
My utility [uhubctl](https://github.com/mvp/uhubctl) is broken on Mac OS Sierra 10.12.4 (16E195) because `libusb_get_port_numbers()` always returns just 1 as list of port numbers in this call:
https://github.com/mvp/uhubctl/blob/master/uhubctl.c#L346

    int pcount = libusb_get_port_numbers(dev, port_numbers, MAX_HUB_CHAIN);

`pcount` returned is always `1` no matter which device it is called for. On older MacOS, it would return unique port_numbers of varying size, even running on exact same hardware.

This may be somehow related to #127.

mvp, 2017-06-01:
Hi, can anyone from libusb team take a look at this? This is extremely annoying that `libusb_get_port_numbers()` does not return device topology :(

mvp, 2017-06-01:
This seems to be fixed in 871eb299b9. Can we please have next libusb release?
OR, perhaps make sure that this is fixed in brew...

hjelmn, 2017-06-01:
I agree. We will start discussing the next release soon.

mvp, 2017-06-06:
Any ETA for next release? Not many homebrew users are technical enough to build libusb from HEAD to fix this issue :(

mvp, 2017-06-16:
Any update on new release? libusb is broken on Mac OS Sierra :(

mvp, 2017-06-28:
@hjelmn, any news on next release?

lame-turtlez, 2017-06-28:
@hjelmn  we really need a new and working release. This is completely broken for 10.12 . Thank you for your hard work.,

mvp, 2017-08-16:
@hjelmn, what are the expectations of having 1.0.22 release cut anytime soon?
According to [libusb 1.0.22 milestone](https://github.com/libusb/libusb/milestone/6), libusb 1.0.22 should have been released 2 weeks ago...

hjelmn, 2017-08-19:
Sorry it took so long to get back to this. I will see where we are now. There are a couple things that were desired for 1.0.22 but I don't know what has been completed.

mvp, 2017-09-26:
Any progress on getting 1.0.22 released?
It's been stalled in limbo for quite a while now...

hjelmn, 2017-09-26:
Sorry about that. Was planning to do a release last month but there were multiple outstanding Windows issues that needed to be resolved. I have gotten tied up with travel the last couple of weeks but will be willing to take another look next week.

dickens, 2017-09-27:
I've nearly finished resolving the Windows issues. There are multiple fixes that will improve many aspects of the Windows backend.

mvp, 2017-10-09:
This problem is open for 5 months now. Can we just have release already? Sorry for being so impatient, but this is a lot of time to wait for simple release of what was fixed long time ago.

mvp, 2017-12-04:
Full 6 months later, [libusb 1.0.22 milestone](https://github.com/libusb/libusb/milestone/6) is past due once again, and there is no evidence that blocking issues for it are going to be fixed anytime soon.

Can you please just cut 1.0.22 as is from trunk?

mvp, 2018-03-12:
@hjelmn, @dickens: as far as I can tell, libusb-1.0.22-rc3 was out for a while, and it was supposed to be last release candidate.
Can you guys release libusb-1.0.22 now?

dickens, 2018-03-25:
v1.0.22 has been released, closing this issue.

closed 2018-03-25 by dickens