libusb/libusb #305: Windows 10: libusb_control_transfer to HID - Input/output error

yaroslavVl, 2017-05-26:
Qt creator 5.8+MinGW (box)  + libusb 1.0.21
**Host machine: win10 x64** (i7-3630 2.4GHz (not very old))))
Device: HID (cnc pendant WHB04B)

r = libusb_interrupt_transfer(handle, pendant_endpoint_in, buf, sizeof(buf), &ret, 100);
goes OK (wait data from HID)
_r = libusb_control_transfer(handle, 0x21, 0x09, 0x0306, 0, data1, 8, 0);_
(sent data to HID)
Input/output error (r : -1)

BUT!: it works (real works - data come to the HID and are displayed correctly)
on linux debian 8 (other machine), raspbian (Raspberry pi3 (ARM)),
and linux debian 8 virtual machine (oracle vm) on the **same machine**

mcuee, 2017-09-10:
Maybe you want to try HIDAPI instead for HID device.

mcuee, 2018-04-16:
Close this one as it is too old and no new info were provided by the OP.

closed 2018-04-16 by mcuee

zangetsudo, 2020-11-11:
hi, robomechs 
im facing the same issue with:
1. v1.0.23: libusb 1.0.23, 
2. linux: gcc version 5.3.1 20160113 (Linaro GCC 5.3-2016.02) 
3. kernel 4.4.84
have you fixed it ?