libusb/libusb #307: 'Device busy' errors with gmtp under TrueOS

danboid, 2017-05-28:
I'm getting multi-minute delays between operations when trying to use gmtp under TrueOS to transfer files to/from my LG G4 phone running stock Android 6. I first reported this to to TrueOS and the gmtp maintainer but he believes the issue most likely lies within libusb or the related FreeBSD kernel code.

I've given a more detailed description of my problem and lots of debug logs from libmtp and gmtp here:

https://github.com/trueos/trueos-core/issues/1426

Thanks

danboid, 2017-05-28:
Note that MTP works fine with my phone under Linux and Windows. I only get these delays under TrueOS.

jbeich, 2017-08-22:
FreeBSD provides its own implementation of libusb with compatible API (0.1, 1.0, 2.0). This one isn't supported. So, try filing a bug assigned to freebsd-usb@FreeBSD.org mailing list.
```
$ ./autogen.sh
$ ./configure
[...]
checking operating system... configure: error: unsupported operating system amd64-unknown-freebsd12.0
```

mcuee, 2017-09-10:
Will close this since it is not a libusb bug.

closed 2017-09-10 by mcuee