libusb/libusb #313: I want to use this lib on Mac Pro~~ .Help me!!

rain3k, 2017-06-15:
I have two computer (Mac Book Pro, Mac Pro)

I want to receive data from two Kinect on one machine. so I developed a program which use libfreenect on Mac Book Pro(13-inch Retina Early 2015/macOS Sierra).

but I want to use my Mac Pro. so..
copied my program to my Mac Pro, and run....

But... my program didn't receive data from two Kinect on Mac Pro.

my program stuck this code (listener.waitForNewFrame(frames);)

this is log....  plz help me.....
libusb: warning [darwin_transfer_status] transfer error: unknown error (0xe000400e) (value = 0xe000400e)

hjelmn, 2017-08-19:
That is

#define kIOUSBNotSent1Err       iokit_usb_err(14)  // 0xe000400e Transaction not sent

Should work the same in Mac Pro as MacBook Pro. What OS version is running on the Mac Pro?

mcuee, 2017-10-07:
Will close this if no further reply in one month time.

closed 2017-10-19 by mcuee