libusb/libusb #314: OpenBSD: libusb_get_port_number is unsupported

vuokko, 2017-06-16:
As far as I can tell there isn't any support.

A quick look in OpenBSD source might provide way to make it work.

1. Start with root hub ioctl GET_DEVICEINFO
2. udi_nports and udi_ports[16] are there for hub
3. ioctl all devices
4. Build the path
5. Return the info.

But I don't know how to implement it right way.

mcuee, 2017-09-10:
We rely on OpenBSD developers on this issue. You can ask in OpenBSD related development group to see if someone there can help.

mcuee, 2017-10-10:
Set to wontfix as well if no OpenBSD developers jump in.

dickens, 2017-12-28:
Since no OpenBSD developer has responded, I'm going to close this issue.

closed 2017-12-28 by dickens

colinhb, 2018-10-31:
For posterity, this was fixed in the OpenBSD port - assume that it will make its way upstream. See CVS log: [http://cvsweb.openbsd.org/cgi-bin/cvsweb/ports/devel/libusb1/patches/patch-libusb_os_openbsd_usb_c?r1=1.7](http://cvsweb.openbsd.org/cgi-bin/cvsweb/ports/devel/libusb1/patches/patch-libusb_os_openbsd_usb_c?r1=1.7).

commit 96898a25 (2020-08-10): OpenBSD: support libusb_get_port_number

From mpi@openbsd.org

Fixes #314
Closes #764