libusb/libusb #316: changing usb host clock time, is this possible?

timprepscius, 2017-06-20:
Greetings,

I am researching an issue I am having with a kinect.  There is no "problem" per say, but I am attempting to perhaps do something outside of the spec.

I have an osx computer, which has a large monotonic clock skew.  Meaning, not only does delta std::chrono::high_resolution clock/monotonic clock skew away from system clock (which is probably being corrected via some ntp-ish thing), it skews away from a known "good clock."  (although etc etc, what is really real??)

If I hook up a kinect to this computer.  I believe it produces 30fps.  In relation to this skewing monotonic clock.

If I hook up the kinect to a different computer, which doesn't have the large skew, it also produces 30fps, in relation to its non-skewing monotonic high resolution clock.

This I what I believe I am seeing.  Perhaps I will find out I am gravely mistaken, however, at this point I'm going on the data I have collected and just watching it.

----

Here is my theory for what is happening.  Does this make sense?

1.  The usb host controller is getting some time interrupt from the system.  Every X nano seconds.  Maybe every 125 nano seconds.

2.  Every time it gets one of these "clicks" it sends a Start-of-Frame packet (SOF) to the Kinect.

3.  The kinect then responds with a packet (or not).  And increments time by 125 nano seconds.

By this mechanism, the kinect, where as physical time on both systems is not actually 125 nano seconds, maybe on one it is 124.8 and one it is 125.2.  Still syncs with each system.

--------

Does this make sense???

If so.  Is it possible to somehow override this usb host clock time?  Make it so it takes skew into account?  I want to somehow control exactly when the kinect believes 33.333etc milliseconds has passed.  I am trying to perfectly align kinect pictures.

I can use any OS.  I'm using osx right now cause it is convenient, but I can change to anything that might provide this functionality.

--

Thank you for any information.

timrprobocom, 2017-08-18:
No.  Remember, your USB host controller is a piece of hardware.  It has to keep working whether there happens to be an operating system or not.  It can't rely on anything as ephemeral as interrupts.

Your host controller generates its own 480 MHz master clock, either by a crystal or a PLL.  That clock is used to generate the bit train on the cable, and all the devices sync themselves to it.  The 125us timing comes from counting those 480 MHz pulses.  It is independent of the motherboard clock, or the operating system timing.  The host controller driver will line up a schedule of transfers to happen in the next microframe, but it's up to the hardware to know when the microframe starts so it can start processing them.

The Kinect also has its own clock source generating a pixel clock, one tick for each pixel.  When it has empties all the pixels from the sensor, it marks the packet as "ready" and wait for a USB read request.  The sensor is configured so that the frame takes about 33ms to complete.  Again, the operating system timing is not involved.  The Kinect doesn't do any 125us timing.  It, like all USB devices, is a slave to the host controller.  When it has data ready to send, it waits for an "IN" token from the host controller.  It doesn't know how long it's going to wait.

None of that can be changed.  It's all hardware.

dickens, 2017-08-19:
I don't think there's anything for libusb here, so I'm closing this issue. Let me know otherwise.

closed 2017-08-19 by dickens