libusb/libusb #317: libusb return LIBUSB_CANCEL_ERROR error when using interrupt transfer in mac os

tluson, 2017-07-04:
I'm using libusb library in my mac,  OS 10.11. I claim four endpoint, two for bulk type and another two for interrupt type. but, the interrupt endpoint transfer callback function return transfer cancel error in mac os, while the bulk callback function is okay. And with all the same code, there is not such error in windows 7 system. Here is my test code.

[listdevs.txt](https://github.com/libusb/libusb/files/1121168/listdevs.txt)

dickens, 2017-12-28:
Can you provide a debug log? There might be some race going on with timeout handling since the OS doesn't handle timeouts for interrupt transfers.

dickens, 2018-03-25:
No response, so closing this for now. The OP can reopen if there is additional information to provide.

closed 2018-03-25 by dickens