libusb/libusb #322: Add CMakeLists.txt file.

sahil56, 2017-07-12:
I wish to create _CMakeLists.txt_ file for this project. Although VS solution files are provided, one might to create project files for other IDEs e.g. Code Blocks for building.

hjelmn, 2017-08-19:
Don't think this is going to happen. It was discusses at length some time ago and it was determined that cmake is a solution in search of a problem. We already provide VS and Xcode projects.

Youw, 2023-08-06:
For reference, help those who are trying to google this one: now there is a community-supported https://github.com/libusb/libusb-cmake

seanm, 2026-09-25:
I'd love cmake support, and apparently so would @Youw who maintains that fork.  That's at least 2 frequent/active libusb contributors.

Why can't we have CMake support in mainline?

Youw, 2026-11-05:
I guess at this point - only to keep main development branch out of burden to support two build systems at the time. Unlike you and me - not all are as happy about it, and I can understand them without agreeing.

mcuee, 2026-11-06:
I am fine to add CMake myself.

@tormodvolden and @sonatique 
Just wondering if you are okay to add CMake to libusb. Thanks.

seanm, 2026-11-06:
> only to keep main development branch out of burden to support two build systems at the time.

There's already more than two: there's an Xcode folder, an msvc folder, and of course autotools.  So that's 3 today.  They could probably all be removed in favour of cmake.

Youw, 2026-11-07:
> They could probably all be removed in favour of cmake.

Maybe not in one go. One thing is to have a new project and claim only CMake support. Totally differetn thing is when there is an existing build systems and I'm sure there are MSVC/XCode project, that include libusb directly as corresponding subprojects, and switching to CMake would require additional step to generate corresponding MSVC/XCode projects first before those could be included, and that is a manual step  that is not always easy to integrate.

But sure, long term we can deprecate and eventually obsolete those.

seanm, 2026-11-14:
>But sure, long term we can deprecate and eventually obsolete those.

Yeah, I meant long term.