libusb/libusb #323: Add Conan support

uilianries, 2017-07-16:
Hi!

Conan is a create tool for package management, it's different from other packages, as .rpm or .dpkg because it can export, build, test and deploy the project.

I created some support for libusb [here](https://github.com/uilianries/conan-libusb), but would be nice if the project provides it own recipe, as result, anyone could obtain libusb on any platform.

I can help libusb project to develop such feature.

LudovicRousseau, 2017-07-28:
libusb does not provide configuration files to generate ``.rpm``or ``.deb`` files because these configuration files are better managed by RPM and Debian maintainers using there own tools.

I did not know Conan. I think it is best if you manage the Conan support files in your own repository.

uilianries, 2017-07-28:
I appreciated your explanation. 

Thanks!

closed 2017-07-28 by uilianries