libusb/libusb #324: Data lost due to timeout of data showing in hardware protocol analyser

Proclivis, 2017-07-24:
This behavior is the same as Ticket #101 in the old track system.

A BulkWrite followed by a BulkRead fails due to a timeout. The read data is present in a protocol trace using a Total Phase Beagle, but not showing in usbmon.

I am providing these files:

ReadFail.tdc (Beagle file you can open with Total Phase Data Center)
ReadFail.txt (A CSV of the Beagle file)
USBError.txt (Top of file has usbmon data with Indexes to Beagle trace, then libusb log with similar annotations.

Basic information:

Device is FX3 (USB 2.0 version)
Dell 9550 with Ubuntu 16
/usr/local/lib/libusb-1.0.so.0.1.0 (Compiled 1.0.20 sources)
Using NuGet package LibUsbDotNet

After this fail, I downloaded 1.0.21 for testing. Assuming LD_LIBRARY_PATH still works...

I get the same exact results.

Final note, I have the same testing application running on Windows using the Cypress driver, and it works properly. This mostly just demonstrates no issues with my FX3 firmware.
[USBError.txt](https://github.com/libusb/libusb/files/1170720/USBError.txt)

Proclivis, 2017-07-24:
Adding files.

[FailedRead.csv.gz](https://github.com/libusb/libusb/files/1170728/FailedRead.csv.gz)
[FailedRead.tdc.gz](https://github.com/libusb/libusb/files/1170729/FailedRead.tdc.gz)

diabolo38, 2017-07-24:
why do you keep opening/closing the device  + do a few get device desc ? 

i can see some strange  message and call in midlle of the  xfer  with issue 

[ 0.407408] [00001545] libusb: debug [libusb_submit_transfer] transfer 0x127a798
[ 0.407411] [00001545] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
***** Index 240 in Beagle shows 9 bytes returned *****
[ 0.407414] [00001545] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 64
_[ 0.913973] [00001560] libusb: debug [libusb_get_device_list] 
[ 0.914015] [00001560] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.914033] [00001560] libusb: debug [libusb_get_device_descriptor] 
[ 0.914042] [00001560] libusb: debug [libusb_get_device_descriptor] 
[ 0.914049] [00001560] libusb: debug [libusb_get_device_descriptor] 
[ 0.914108] [00001560] libusb: debug [libusb_get_device_descriptor] 
[ 0.914124] [00001560] libusb: debug [libusb_get_device_descriptor] 
[ 0.914139] [00001560] libusb: debug [libusb_get_device_descriptor] 
[ 0.914151] [00001560] libusb: debug [libusb_get_device_descriptor] 
[ 0.914165] [00001560] libusb: debug [libusb_get_device_descriptor] 
[ 0.914181] [00001560] libusb: debug [libusb_get_device_descriptor] 
[ 0.914194] [00001560] libusb: debug [libusb_get_device_descriptor]_ 
[ 1.407501] [0000154e] libusb: debug [handle_events] poll() returned 1
[ 1.407534] [0000154e] libusb: debug [handle_events] timerfd triggered
[ 1.407549] [0000154e] libusb: debug [libusb_cancel_transfer] transfer 0x127a798
[ 1.407631] [0000154e] libusb: debug [disarm_timerfd] 

why is this "get dev list"  hapening in middle of  on going bulk xfer ?
even that maybe ok that is quite unusual way to proceed 

the beagle capture clearly show that the last bulk in xfer is not reaching libusb

Proclivis, 2017-07-24:
There is one open/close of the driver to check if there is a need to download firmware:

```csharp
        public override Boolean Open()
        {
            try
            {
                // Force the use of libusb on windows.
                UsbDevice.ForceLibUsbWinBack = true;
                MyUsbDevice = UsbDevice.OpenUsbDevice(MyUsbFinder);
                if (MyUsbDevice == null) throw new Exception("Device Not Found.");

                IUsbDevice wholeUsbDevice = MyUsbDevice as IUsbDevice;
                wholeUsbDevice.SetConfiguration(1);
                wholeUsbDevice.ClaimInterface(0);
                reader = MyUsbDevice.OpenEndpointReader(ReadEndpointID.Ep01);
                writer = MyUsbDevice.OpenEndpointWriter(WriteEndpointID.Ep01);
            }
            catch (Exception ex)
            {
                return false;
            }
            return true;
        }

        public override void Close()
        {
            if (MyUsbDevice != null) 
                {
                    if (MyUsbDevice.IsOpen)
                    {
                        IUsbDevice wholeUsbDevice = MyUsbDevice as IUsbDevice;
                        if (!ReferenceEquals(wholeUsbDevice, null))
                        {
                            wholeUsbDevice.ReleaseInterface(0);
                        }
                        MyUsbDevice.Close();
                    }
                    MyUsbDevice = null;
                }
        }
```

Then one open close for the test:

```
        public DongleIOLibUsb()
        {
            UsbDeviceNotifier.OnDeviceNotify += OnDeviceNotifyEvent;
            UsbDevice.UsbErrorEvent += OnErrorNotifyEvent;

            try {
                // Force the use of libusb on windows.
                UsbDevice.ForceLibUsbWinBack = true;

                UsbDeviceFinder finder = new UsbDeviceFinder(0x0456, 0xB310);
                MyUsbDevice = UsbDevice.OpenUsbDevice(finder);

                IUsbDevice wholeUsbDevice = MyUsbDevice as IUsbDevice;
                if (!ReferenceEquals(wholeUsbDevice, null))
                {
                    wholeUsbDevice.SetConfiguration(1);
                    wholeUsbDevice.ClaimInterface(0);
                    FrameIn = wholeUsbDevice.OpenEndpointReader(ReadEndpointID.Ep01);
                    FrameOut = wholeUsbDevice.OpenEndpointWriter(WriteEndpointID.Ep01);
                    PacketIn = wholeUsbDevice.OpenEndpointReader(ReadEndpointID.Ep02);
                    PacketOut = wholeUsbDevice.OpenEndpointWriter(WriteEndpointID.Ep02);

                }
            }
[... 66 lines cut ...]
[ 1.405744] [000048e0] libusb: debug [handle_events] poll() returned 1
[ 1.405777] [000048e0] libusb: debug [handle_events] timerfd triggered
[ 1.405796] [000048e0] libusb: debug [libusb_cancel_transfer] transfer 0x1f7f2b0
```
There is still a getting of descriptors and a cancel. So the cancel is probably not associated with Abort.

I wonder if LibUsbDotNet has some code that fetches descriptors when there are errors?

Another test I did was to copy the test code and run it twice in a row while open, and it works all the time. And you get working messages like:

```
[ 0.554150] [00004af5] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.554160] [00004af5] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 64
[ 0.564267] [00004afe] libusb: debug [handle_events] poll() returned 1
[ 0.564310] [00004afe] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=8
[ 0.564317] [00004afe] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 0.564323] [00004afe] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
```
And there is no getting of descriptors. This leads back to LibUsbDotNet behavior.

So I guess it is hard to be certain if the getting of descriptors is a cause or an effect.

From the LibUSbDotNet's perspective, it is returning IOTimedOut. The libusb log indicates that it is getting descriptors after each timeout. Maybe LibUsbDotNet has some strategy to check endpoints to make sure they did not go away after a timeout.

Any ideas on how to narrow this down? LibUsbDotNet may not be all that easy to compile and debug in my larger environment.

Proclivis, 2017-07-24:
I managed to compile LibUsbDotNet. There is a polling loop in a background thread getting the descriptors. Disabling it does not make the problem go away.

Ultimately we get to:

```csharp
return (MonoUsbError)MonoUsbApi.SubmitTransfer(handle);

        [DllImport(LIBUSB_DLL, CallingConvention = CC, SetLastError = false, EntryPoint = "libusb_submit_transfer")]
        internal static extern int SubmitTransfer(IntPtr pTransfer);
```
Leading to:

```
[121.563624] [00004602] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[121.563635] [00004602] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 64
[122.082080] [00004612] libusb: debug [handle_events] poll() returned 0
```
One step in the debugger to return and update the display:

```
[144.472392] [00004612] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[144.472425] [00004612] libusb: debug [handle_events] poll() 3 fds with timeout in 2000ms
[144.472438] [00004612] libusb: debug [handle_events] poll() returned 1
[144.472442] [00004612] libusb: debug [handle_events] timerfd triggered
[144.472455] [00004612] libusb: debug [libusb_cancel_transfer] transfer 0xa77020
[144.472531] [00004612] libusb: debug [disarm_timerfd] 
[144.472556] [00004612] libusb: debug [handle_events] poll() 3 fds with timeout in 0ms
[144.472567] [00004612] libusb: debug [handle_events] poll() returned 1
[144.472579] [00004612] libusb: debug [reap_for_handle] urb type=3 status=-2 transferred=0
[144.472587] [00004612] libusb: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
[144.472593] [00004612] libusb: debug [handle_bulk_completion] abnormal reap: urb status -2
[144.472599] [00004612] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
[144.472606] [00004612] libusb: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
[144.472612] [00004612] libusb: debug [disarm_timerfd] 
[144.472618] [00004612] libusb: debug [usbi_handle_transfer_completion] transfer 0xa77020 has callback 0x7f3a1ddd30fc
[144.472798] [00004612] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[144.472815] [00004612] libusb: debug [handle_events] poll() 3 fds with timeout in 2000ms
[146.474994] [00004612] libusb: debug [handle_events] poll() returned 0
```

The timeout status ultimately comes from code peeking in a packet:

```csharp
        public MonoUsbTansferStatus Status
        {
            get { return (MonoUsbTansferStatus)Marshal.ReadInt32(mpMonoUsbIsoPacket, OfsStatus); }
            set { Marshal.WriteInt32(mpMonoUsbIsoPacket, OfsStatus, (int)value); }
        }
```

Seems like the descriptor is not the problem, and the problem all occurs in libusb between a transfer call and a packet returned, unless LibUsbDotNet is mangling the data.

Proclivis, 2017-07-24:
Anyone have some clarity on the original statement: the beagle capture clearly show that the last bulk in xfer is not reaching libusb

What I see is a valid IN txn at index 239 followed by an IN-NAK poll failure 2s later. I am guessing libusb or LibUsbDotNet issues this poll in response to the original timeout.

From looking at the libusb code, and seeing that -2 is timeout, it just looks to me that the IN transaction goes over the USB, is ACKed by the USB master, but somehow becomes a timeout.

Proclivis, 2017-07-25:
The libusb_cancel_transfer does not always occur. Here is a case where I put a breakpoint on C# and proved there is no cancel call to libusb.

```
[13.020782] [0000734a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[13.020815] [0000734a] libusb: debug [handle_events] poll() 3 fds with timeout in 2000ms
[13.021836] [00007338] libusb: debug [libusb_alloc_transfer] transfer 0x19db620
[13.021891] [00007338] libusb: debug [libusb_submit_transfer] transfer 0x19db620
[13.021898] [00007338] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[13.021909] [00007338] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 64
[14.022066] [0000734a] libusb: debug [handle_events] poll() returned 1
[14.022096] [0000734a] libusb: debug [handle_events] timerfd triggered
[14.022110] [0000734a] libusb: debug [libusb_cancel_transfer] transfer 0x19db620
[14.022167] [0000734a] libusb: debug [disarm_timerfd] 
[14.022178] [0000734a] libusb: debug [handle_events] poll() 3 fds with timeout in 0ms
[14.022186] [0000734a] libusb: debug [handle_events] poll() returned 1
[14.022193] [0000734a] libusb: debug [reap_for_handle] urb type=3 status=-2 transferred=0
[14.022198] [0000734a] libusb: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
[14.022202] [0000734a] libusb: debug [handle_bulk_completion] abnormal reap: urb status -2
[14.022206] [0000734a] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
[14.022210] [0000734a] libusb: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
[14.022214] [0000734a] libusb: debug [disarm_timerfd] 
[14.022218] [0000734a] libusb: debug [usbi_handle_transfer_completion] transfer 0x19db620 has callback 0x7fc6c74230fc
[14.022262] [0000734a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[14.022267] [0000734a] libusb: debug [handle_events] poll() 3 fds with timeout in 2000ms
```

diabolo38, 2017-07-25:
for info the number of nak will varies depending how fast device get ready to respond but the transfer end up correctly on bus , but libusb only see "time out completion"  what is not normal.

I do similar thing with native api in window/linux without problems (sync and assync xfer)
I remember of completion issue when mixing  sync/assynch transfer  without proper race free flags usage in  "libusb_handle_events_completed call"
maybe this is what happening  in libubsdonet  ep reader/writer thread   event handling ?
 i can't help more on that maybe ask support to libusbdonet  team. 
Maybe  the doc http://libusb.sourceforge.net/api-1.0/mtasync.html could be helpfull.

Proclivis, 2017-07-31:
After reviewing the LibUSbDotNet code, what it is doing is:

Thead A is in a loop calling libusb_handle_events_timeout.

Thread B submits a transfer with a callback, then waits on an event. The callback does a Set() on the event, and then the results are processed by Thread B. 

If there is a second endpoint (not my case), there is another event/callback pair.

If the application where multi-threaded, with one thread per endpoint, the callback would have to be well behaved under the condition that multiple submits are called from different threads, and processing completion as well, but always by the same thread that submitted.

Back to my case, after the callback, Thread A sees a Timeout, even though there is data showing in the Beagle. I conclude the behavior is contained withing libusb, and is not a problem in the LibUsbDotNet library.

dickens, 2017-08-25:
There is no indication in any of the logs, especially usbmon, that the host actually received any data from the device. The fact that the urb status is being reported as -2 (ENOENT) means that the urb is being fully cancelled, in other words it was still linked and in progress.

Perhaps there is an issue on the device side such that the data isn't being correctly programmed to be sent out. This does **not** appear to be a libusb issue.

dickens, 2018-03-25:
No response after more than 6 months, so I will close this for now. The OP can reopen if there is additional information to provide.

closed 2018-03-25 by dickens