libusb/libusb #326: Shutdown race with hotplug on Darwin

johnkeeping, 2017-07-27:
I have seen a crash with the stack trace:

```
0   libusb-1.0.0.dylib                0x000000010b4f6556 darwin_deref_cached_device + 38
1   libusb-1.0.0.dylib                0x000000010b4f50f3 darwin_destroy_device + 47
2   libusb-1.0.0.dylib                0x000000010b4ee7c1 libusb_unref_device + 109
3   libusb-1.0.0.dylib                0x000000010b4f0160 libusb_exit + 381
```

Now in this case we are calling `libusb_exit()` in response to a device being unplugged, so I think what's happening is that the `libusb_unref_device()` in `libusb_exit()` is racing with the one in `darwin_devices_detached()` since the latter is not under `usb_devs_lock`.

This is possible because the event loop on Darwin is not shutdown until `darwin_exit()` is called later in `libusb_exit()` as `usbi_backend.exit`, so hotplug events can still arrive at the point where `usb_devs` is cleared.

I'm opening this as an issue rather than a PR because I'm not sure what is the best way to fix this.  Is it possible to just move the loop to clear `usb_devs` to after `usbi_backend.exit` has been called?

seanm, 2019-04-12:
Does this still occur?  If you try building with Thread Sanitizer you might be better able to catch any races.

mcuee, 2020-12-07:
@johnkeeping Does this still occur with the latest version of libusb (say 1.0.24rc2)?

johnkeeping, 2020-12-07:
I don't have the test setup for this any more, so I can't say for sure.  But #701 sounds a lot like this issue!

Since I can't reproduce it now, I'm happy to close this.

closed 2020-12-07 by johnkeeping