libusb/libusb #328: ~ Help code::block errors ~ undefined reference ?

USB-NOOB, 2017-07-29:
Please Help

I am a complete novice so please start with the simplest solution.

Do I need to install something or just use the header file ?

Windows 10 platform

Used #include "libusb.h" in code with file in project folder.  (changed .h to .txt to attach)
[libusb.txt](https://github.com/libusb/libusb/files/1184835/libusb.txt)

The Main Program: (changed .cpp to .txt to attach)
[main.txt](https://github.com/libusb/libusb/files/1184837/main.txt)

Errors:

||=== Build: Debug in USB-test (compiler: GNU GCC Compiler) ===|
obj\Debug\main.o||In function `main':|
C:\-----  Herbs  -----\USB-test\main.cpp|14|undefined reference to `libusb_init@4'|
C:\-----  Herbs  -----\USB-test\main.cpp|19|undefined reference to `libusb_set_debug@8'|
C:\-----  Herbs  -----\USB-test\main.cpp|20|undefined reference to `libusb_get_device_list@8'|
C:\-----  Herbs  -----\USB-test\main.cpp|29|undefined reference to `libusb_free_device_list@8'|
C:\-----  Herbs  -----\USB-test\main.cpp|30|undefined reference to `libusb_exit@4'|
obj\Debug\main.o||In function `Z8printdevP13libusb_device':|
C:\-----  Herbs  -----\USB-test\main.cpp|36|undefined reference to `libusb_get_device_descriptor@8'|
C:\-----  Herbs  -----\USB-test\main.cpp|46|undefined reference to `libusb_get_config_descriptor@12'|
C:\-----  Herbs  -----\USB-test\main.cpp|66|undefined reference to `libusb_free_config_descriptor@4'|
||error: ld returned 1 exit status|
||=== Build failed: 9 error(s), 0 warning(s) (0 minute(s), 0 second(s)) ===|

diabolo38, 2017-07-29:
you have to add the dll or lib to linker option additional lib adn it shall do

dickens, 2017-08-25:
As mentioned, you need to link against the libusb .dll or .lib. How you do this is specific to your build toolset.

closed 2017-08-25 by dickens