libusb/libusb #331: Reading 10+ bulk devices at the same time causes data loss

corby, 2017-08-16:
We are building an industrial appliance that can have up to 100 usb connections at the same time spread out over 5 xHCI controllers.

When I connect one device and do a libusb_bulk_transfer (sync) everything works fine and I get a good data rate.
However, if I try to read from 3 devices at the same time from the same program from 3 different threads, i start loosing packets from the libusb layer.

I'm using GOLANG with an interface layer to native libusb on Centos7.
Each thread has it's own CTX and the same thread is reading from one usb device at one time.
That one thread read with a TO of 0 so it blocks.  when it triggers it reads in the data and places it on a queue to be handled by a different thread, then it goes back to reading the device and blocking.

When I'm reading from 3 or more devices, within a second I drop data on one of the devices and I panic because the data wouldn't parse.
I get no error from libusb and the logs don't show that a packet was received and dropped.

When I introduce a 5ms delay between reads things are clean again, but it slows down my read which isn't good when I'm trying to dump 40MB of data per device.

here is the libusb log:

[64.956500] [0000194b] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[64.956503] [0000194b] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[64.956506] [0000194b] libusb: debug [usbi_handle_transfer_completion] transfer 0x1869c30 has callback 0x7f08f79a8920
[64.956508] [0000194b] libusb: debug [sync_transfer_cb] actual_length=1190
[64.956512] [0000194b] libusb: debug [libusb_free_transfer] transfer 0x1869c30
[64.956554] [0000194b] libusb: debug [libusb_alloc_transfer] transfer 0x1869c30
[64.956560] [0000194b] libusb: debug [libusb_submit_transfer] transfer 0x1869c30
[64.956562] [0000194b] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4096
[64.956570] [0000194b] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[64.956572] [0000194b] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[64.957548] [0000194b] libusb: debug [handle_events] poll() returned 1
[64.957560] [0000194b] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=448
[64.957564] [0000194b] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[64.957566] [0000194b] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[64.957569] [0000194b] libusb: debug [usbi_handle_transfer_completion] transfer 0x1869c30 has callback 0x7f08f79a8920
[64.957572] [0000194b] libusb: debug [sync_transfer_cb] actual_length=448
[64.957575] [0000194b] libusb: debug [libusb_free_transfer] transfer 0x1869c30
[64.957623] [0000194b] libusb: debug [libusb_alloc_transfer] transfer 0x1869c30
[64.957629] [0000194b] libusb: debug [libusb_submit_transfer] transfer 0x1869c30
[64.957632] [0000194b] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4096
[64.957639] [0000194b] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[64.957642] [0000194b] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[64.963367] [0000194b] libusb: debug [handle_events] poll() returned 1
[64.963373] [0000194b] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=1995
[64.963376] [0000194b] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[64.963378] [0000194b] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[64.963381] [0000194b] libusb: debug [usbi_handle_transfer_completion] transfer 0x1869c30 has callback 0x7f08f79a8920
[64.963383] [0000194b] libusb: debug [sync_transfer_cb] actual_length=1995
[64.963389] [0000194b] libusb: debug [libusb_free_transfer] transfer 0x1869c30
[64.963433] [0000194b] libusb: debug [libusb_alloc_transfer] transfer 0x1869c30
[64.963436] [0000194b] libusb: debug [libusb_submit_transfer] transfer 0x1869c30
[64.963439] [0000194b] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4096
[64.963446] [0000194b] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[64.963448] [0000194b] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[64.965616] [0000194b] libusb: debug [handle_events] poll() returned 1
[64.965632] [0000194b] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=57
[64.965637] [0000194b] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[64.965641] [0000194b] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[64.965645] [0000194b] libusb: debug [usbi_handle_transfer_completion] transfer 0x1869c30 has callback 0x7f08f79a8920
[64.965650] [0000194b] libusb: debug [sync_transfer_cb] actual_length=57
[64.965656] [0000194b] libusb: debug [libusb_free_transfer] transfer 0x1869c30

The missing data is between the 448 byte transfer and the 1995 byte transfer and totals 357 bytes that just dissapeared

Any help is appreciated.

timrprobocom, 2017-08-18:
The only way to maximize throughput is by using asynchronous processing, to submit multiple requests.  When a transfer completes, there needs to be another read already queued up.  Otherwise, the host controller won't even ask your device for data during that frame.  Does your device have very small FIFOs?  Is it possible the 357 bytes were caused by FIFO overflow because you didn't get around to reading again quickly enough?

By the way, to get maximum throughput, you need to be transferring full packets.  If you are really sending 448 bytes, that ends the transfer, and you won't get another chance until the next microframe.

dickens, 2017-08-25:
I agree with Tim, there is likely data being lost on the device side. Can you try this with at least two transfers and report whether the problem still occurs? I'll keep this a bit longer for more info.

corby, 2017-08-25:
Actually I've easily been able to replicate this with one device.
I uploaded a known file into the device flash and then tired to download it w/o the 5ms delay between libusb_bulktransfer calls and I loose data.
The device log says it sent the data out through the buffer (though it may still be a device side issue).

I'm now implementing the async method to see if that fixes the problem.

corby, 2017-08-26:
I added the async interface to the golang library and I'm not having any read issues anymore.
You may want to add docs to the sync methods indicating that data loss may occur on high speed connections.

I consider this issue closed.
Resolution: use async libusb_submit_transfer instead of the sync methods.

closed 2017-08-26 by corby

timrprobocom, 2017-08-26:
Corby Wilson wrote:
>
> I added the async interface to the golang library and I'm not having
> any read issues anymore.
> You may want to add docs to the sync methods indicating that data loss
> may occur on high speed connections.
>

Let's be very clear about this.  The only way you can get data loss with
the synchronous interface is if your device's FIFOs overflow.  With the
synchronous interface, you can't (of course) submit more than one
transfer at a time, so there is always going to be a delay between the
time a bulk transfer is completed, and the time your app can process it
and submit the next bulk transfer.  If your device streams data
continuously and the FIFOs are small, then that delay can certainly be
enough to cause overflow.  However, that is a fundamental USB design
issue.  It is certainly not unique to libusb.  Continuously streaming
devices require asynchronous transfers, so you can submit multiple
transfers and always have an empty buffer waiting.

-- 
Tim Roberts, timr@probo.com
Providenza & Boekelheide, Inc.