libusb/libusb #334: libusb on Windows stops after sending many outgoing packets

savnat-kris, 2017-08-24:
Has anyone ran into a problem with sending a lot of messages on Windows 10 with libusb and getting a lock up over small length of time?

Here are some additional details:
- Windows 10 Pro
- Intel Core i7-6700 @ 3.40 Ghz, 64-bit OS & 32 GB RAM
- Outgoing message (PC->to->USB Device): URB_INTERRUPT
- Outgoing message (PC->to->USB Device): Endpoint 0x02
- Outgoing message (PC->to->USB Device): Send about 1-6 milliseconds apart, thousands of messages
- Outgoing message (PC->to->USB Device): 1024 bytes in size
- Using libusb_fill_interrupt_transfer()/libusb_submit_transfer()

I have an application take takes incoming HID messages and sends out custom messages back to the device as soon as it gets an incoming message (this is on order of low milliseconds).  Running this application for a few thousand messages I am seeing that libusb stops sending messages and receiving messages as well sometimes.  I need to stop the application unplug the USB cable and plug it back in to get it to work again.

Just by adding some debug statements, I can see that when this works great.  I get a libusb_submit_transfer(), followed by a callback with LIBUSB_TRANSFER_COMPLETED.  In cases where it does not work I get a lot of libusb_submit_transfer(), but the call back is delayed significantly and eventually the callback function is called many time one after the other all at once, the status shows LIBUSB_TRANSFER_CANCELLED.

Any help would be much appreciated.

dickens, 2017-08-24:
Can you provide a debug log?

savnat-kris, 2017-08-25:
On Windows libusb uses Microsoft's *OutputDebugString()* function (debugger output and not standard output), so needed to do a few things to redirect the this to the file.  However, I got the data for the failure that occurs under 4 different log levels: (1) ERROR, (2) WARNING, (3) INFO, & (4) DEBUG.

For ERROR, WARNING, & INFO there is nothing in the log files from libusb.  For DEBUG, I can see a lot of libusb entries.  Therefore, I ran the test twice.  The logs are attached.  I needed to compress the DEBUG logs because they are 83 MB and 128 MB in length, so this forum won't allow this large text file upload.

I also tried to see if I can count how many USB messages need to be sent before a problem occurs, but they vary a lot, so it leads me to believe that it is a race condition somewhere.

[freopen.out__LIBUSB_LOG_LEVEL_ERROR.txt](https://github.com/libusb/libusb/files/1252920/freopen.out__LIBUSB_LOG_LEVEL_ERROR.txt)
[freopen.out__LIBUSB_LOG_LEVEL_INFO.txt](https://github.com/libusb/libusb/files/1252919/freopen.out__LIBUSB_LOG_LEVEL_INFO.txt)
[freopen.out__LIBUSB_LOG_LEVEL_WARNING.txt](https://github.com/libusb/libusb/files/1252921/freopen.out__LIBUSB_LOG_LEVEL_WARNING.txt)
[freopen.out__LIBUSB_LOG_LEVEL_DEBUG.zip](https://github.com/libusb/libusb/files/1252932/freopen.out__LIBUSB_LOG_LEVEL_DEBUG.zip)

savnat-kris, 2017-08-31:
The error occurs in **windows_winusb.c** and in the **else** call:

```
	if (IS_XFERIN(transfer)) {
		usbi_dbg("reading %d bytes", transfer->length);
		ret = WinUSBX[sub_api].ReadPipe(wfd.handle, transfer->endpoint, transfer->buffer, transfer->length, NULL, wfd.overlapped);
	} else {
		usbi_dbg("writing %d bytes", transfer->length);
		ret = WinUSBX[sub_api].WritePipe(wfd.handle, transfer->endpoint, transfer->buffer, transfer->length, NULL, wfd.overlapped);
	}

```
The return is ERROR_IO_PENDING 997 (0x3E5) and is described as *Overlapped I/O operation is in progress*.

The subsequent code logic in the function specifically ignores the **ERROR_IO_PENDING** and returns **LIBUSB_SUCCESS**: 
```
	if (!ret) {
		if (GetLastError() != ERROR_IO_PENDING) {
			usbi_err(ctx, "ReadPipe/WritePipe failed: %s", windows_error_str(0));
			usbi_free_fd(&wfd);
			return LIBUSB_ERROR_IO;
		}
	} else {
```
This is why the error is not reported through the standard **libusb** error logs.

savnat-kris, 2017-09-07:
That's the thing... I don't see an obvious error from the logs.

Right now I see a point in time where all the outgoing packets are being cancelled using _libusb_cancel_transfer()_.

savnat-kris, 2017-09-18:
I spent a few days working with someone from Microsoft USB team on this.

We used a USB trace to capture a ETL file while reproducing the issue.  From that trace file we are able to see that the reason the packets stop sending is because a *WinUSB_AbortPipe* call is being made by *libusb* in *winusbx_abort_transfers()*.  The main reason this happens is because all transfers have a timeout value and this timeout value expires, so I decided to remove the *WinUSB_AbortPipe* call by setting the timeout value to 0:
```
int API_EXPORTED libusb_submit_transfer(struct libusb_transfer * transfer)
{
    struct usbi_transfer *itransfer =
        LIBUSB_TRANSFER_TO_USBI_TRANSFER(transfer);
    struct libusb_context *ctx = TRANSFER_CTX(transfer);
    int r;

    // DEBUG Only
    transfer->timeout = 0;
    usbi_dbg("transfer %p", transfer);
...
}
```

After doing this, I see that now I am running out of *windows poll fds*.  There is static memory allocated in poll_windows.h:
```
#define MAX_FDS     256
```

I also used *libusb* logging to see what is happening in the output.
[output_LIBUSB_LOG_LEVEL_DEBUG_PartialEnd.zip](https://github.com/libusb/libusb/files/1311422/output_LIBUSB_LOG_LEVEL_DEBUG_PartialEnd.zip)

It is clear from the log that *libusb* is running out of polling file descriptors overtime.  The *WinUSB_WritePipe* call is using a *Overlapped* pointer to a structure to keep track of asynchronous operations.  There is a function in *windows_winusb.c* that looks at the overlapped results to clear the file descriptors:
```
void windows_get_overlapped_result(struct usbi_transfer *transfer, struct winfd *pollable_fd, DWORD *io_result, DWORD *io_size)
{
	if (HasOverlappedIoCompletedSync(pollable_fd->overlapped)) {
		*io_result = NO_ERROR;
		*io_size = (DWORD)pollable_fd->overlapped->InternalHigh;
	} else if (GetOverlappedResult(pollable_fd->handle, pollable_fd->overlapped, io_size, false)) {
		// Regular async overlapped
		*io_result = NO_ERROR;
	} else {
		*io_result = GetLastError();
	}
}
```
Either the underlying Windows calls are not able to track get the status of these write calls or *libusb* polling method does not work correctly.  I am still attempting to track this down.

@dickens, have you ran into a problem with *HasOverlappedIoCompletedSync()* or *GetOverlappedResult()* not being able to get the status correctly in the past?

savnat-kris, 2017-09-21:
The problem is with the Windows file descriptors not clearing out over time.  The log starts out fine, but then the file descriptors are not getting cleared out.

I took a screen capture (attached) of the log file and marked the parts to show what is happening.
[image]

Also, here is the parts that are highlighted:

**Working:**
```
[ 6.914509] [00001dd4] libusb: debug [windows_handle_events] checking fd 2 with revents = 0000
[ 6.914516] [00001dd4] libusb: debug [windows_handle_events] checking fd 1 with revents = 0004
[ 6.914523] [00001dd4] libusb: debug [usbi_remove_pollfd] remove fd 1
[ 6.914531] [00001dd4] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 0, size 1024
[ 6.914543] [00001dd4] libusb: debug [usbi_handle_transfer_completion] transfer 00794E14 has callback 0F5642D9
[ 6.914550] [00001dd4] libusb: debug [libusb_free_transfer] transfer 00794E14
[ 6.914559] [00001dd4] libusb: debug [windows_handle_events] checking fd 4 with revents = 0001
[ 6.914566] [00001dd4] libusb: debug [usbi_remove_pollfd] remove fd 4
```

**Not Working:**
```
[136.946856] [00001dd4] libusb: debug [windows_handle_events] checking fd 66 with revents = 0000
[136.946862] [00001dd4] libusb: debug [windows_handle_events] checking fd 67 with revents = 0000
[136.946869] [00001dd4] libusb: debug [windows_handle_events] checking fd 68 with revents = 0000
[136.946875] [00001dd4] libusb: debug [windows_handle_events] checking fd 69 with revents = 0000
[136.946881] [00001dd4] libusb: debug [windows_handle_events] checking fd 70 with revents = 0000
[136.946888] [00001dd4] libusb: debug [windows_handle_events] checking fd 71 with revents = 0000
[136.946894] [00001dd4] libusb: debug [windows_handle_events] checking fd 72 with revents = 0000
[136.946901] [00001dd4] libusb: debug [windows_handle_events] checking fd 73 with revents = 0000
[136.946907] [00001dd4] libusb: debug [windows_handle_events] checking fd 74 with revents = 0000
[136.946913] [00001dd4] libusb: debug [windows_handle_events] checking fd 75 with revents = 0000
[136.946920] [00001dd4] libusb: debug [windows_handle_events] checking fd 76 with revents = 0000
[136.946926] [00001dd4] libusb: debug [windows_handle_events] checking fd 77 with revents = 0000
[136.946932] [00001dd4] libusb: debug [windows_handle_events] checking fd 78 with revents = 0000
[136.946938] [00001dd4] libusb: debug [windows_handle_events] checking fd 79 with revents = 0000
[136.946945] [00001dd4] libusb: debug [windows_handle_events] checking fd 80 with revents = 0000
[136.946951] [00001dd4] libusb: debug [windows_handle_events] checking fd 81 with revents = 0000
[136.946958] [00001dd4] libusb: debug [windows_handle_events] checking fd 82 with revents = 0000
[136.946964] [00001dd4] libusb: debug [windows_handle_events] checking fd 83 with revents = 0000
[136.946970] [00001dd4] libusb: debug [windows_handle_events] checking fd 84 with revents = 0000
[136.946989] [00001dd4] libusb: debug [windows_handle_events] checking fd 85 with revents = 0000
[136.946995] [00001dd4] libusb: debug [windows_handle_events] checking fd 86 with revents = 0000
[136.947002] [00001dd4] libusb: debug [windows_handle_events] checking fd 87 with revents = 0000
[136.947008] [00001dd4] libusb: debug [windows_handle_events] checking fd 88 with revents = 0000
[136.947015] [00001dd4] libusb: debug [windows_handle_events] checking fd 89 with revents = 0000
[136.947021] [00001dd4] libusb: debug [windows_handle_events] checking fd 90 with revents = 0000
[136.947027] [00001dd4] libusb: debug [windows_handle_events] checking fd 91 with revents = 0000
[136.947034] [00001dd4] libusb: debug [windows_handle_events] checking fd 92 with revents = 0000
[136.947040] [00001dd4] libusb: debug [windows_handle_events] checking fd 93 with revents = 0000
[136.947047] [00001dd4] libusb: debug [windows_handle_events] checking fd 94 with revents = 0000
[136.947053] [00001dd4] libusb: debug [windows_handle_events] checking fd 95 with revents = 0000
[136.947059] [00001dd4] libusb: debug [windows_handle_events] checking fd 96 with revents = 0000
[136.947065] [00001dd4] libusb: debug [windows_handle_events] checking fd 97 with revents = 0000
[136.947072] [00001dd4] libusb: debug [windows_handle_events] checking fd 98 with revents = 0000
[136.947078] [00001dd4] libusb: debug [windows_handle_events] checking fd 99 with revents = 0000
[136.947085] [00001dd4] libusb: debug [windows_handle_events] checking fd 100 with revents = 0000
[136.947091] [00001dd4] libusb: debug [windows_handle_events] checking fd 101 with revents = 0000
[136.947098] [00001dd4] libusb: debug [windows_handle_events] checking fd 102 with revents = 0000
[136.947104] [00001dd4] libusb: debug [windows_handle_events] checking fd 103 with revents = 0000
[136.947110] [00001dd4] libusb: debug [windows_handle_events] checking fd 104 with revents = 0000
[136.947117] [00001dd4] libusb: debug [windows_handle_events] checking fd 105 with revents = 0000
[136.947123] [00001dd4] libusb: debug [windows_handle_events] checking fd 106 with revents = 0000
[136.947129] [00001dd4] libusb: debug [windows_handle_events] checking fd 107 with revents = 0000
[136.947136] [00001dd4] libusb: debug [windows_handle_events] checking fd 108 with revents = 0000
[136.947142] [00001dd4] libusb: debug [windows_handle_events] checking fd 109 with revents = 0000
[136.947148] [00001dd4] libusb: debug [windows_handle_events] checking fd 110 with revents = 0000
[... cut ...]
[136.947224] [00001dd4] libusb: debug [usbi_handle_transfer_completion] transfer 0070AD64 has callback 0F5641EE
[136.948715] [00001dd4] libusb: debug [libusb_alloc_transfer] transfer 0070B154
[136.948735] [00001dd4] libusb: debug [libusb_submit_transfer] transfer 0070B154
[136.948742] [00001dd4] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 02 with interface 1
[136.948753] [00001dd4] libusb: debug [winusbx_submit_bulk_transfer] writing 1024 bytes
[136.948765] [00001dd4] libusb: debug [usbi_add_pollfd] add fd 116 events 4
```

dickens, 2017-09-21:
The number of "file descriptors" being used appears to be the result of the logic in your application. It looks like you are submitting two transfers at a time, one OUT transfer to EP 0x02 (1,024 bytes) and one IN transfer to EP 0x81 (64 bytes). Your application appears to be submitting two transfers every time the IN transfer completes, but it looks like the device is not keeping up with the rate of OUT transfers and over time they end up stacking up. Perhaps your application should wait for the OUT transfer to complete before submitting another one? I don't know the exact protocol you are implementing, so I can't comment much more than this.

savnat-kris, 2017-10-04:
Ok, let ask something a little bit more basic.

If I want to received something from a device on one endpoint and send data to that same device on a different endpoint, do I need two different USB contexts (libusb_context)?

dickens, 2020-08-18:
> Ok, let ask something a little bit more basic.
> 
> If I want to received something from a device on one endpoint and send data to that same device on a different endpoint, do I need two different USB contexts (libusb_context)?

No, you do not need multiple USB contexts.

This issue is almost 3 years old now.  Is there still an issue within libusb that needs to be addressed?

mcuee, 2020-09-03:
Close this for now. OP can reopen if needed.

closed 2020-09-03 by mcuee