libusb/libusb #335: getting real exclusive access to a device under Windows (10)

Otacon5555, 2017-08-29:
My environment is Windows 10 Home 64 bit latest version (1703 at this moment).
I have an USB pedal that libusb (xusb.exe) sees as a class 0 device with two class 3 (HID) interface.

So, guessing it is actually a HID device, I tried the [hidapi](https://github.com/signal11/hidapi) library, as suggested in the libusb [FAQ](https://github.com/libusb/libusb/wiki/FAQ) ("HIDAPI is recommended for HID device than libusb Windows" - "In general, you may find HIDAPI a better library for HID device").

My findings are described in my (first) post in the hidapi thread ["Option to open hid device with exclusive access"](https://github.com/signal11/hidapi/issues/302).

However, since that approach does not seem promising, I'm guessing if I can achieve my goal with libusb, since the device is not seen as a proper classe 3 (HID) device.

I've found also the mailing list discussion ["How to get exclusive access to an USB device"](https://sourceforge.net/p/libusb-win32/mailman/message/3744467/) and I'm aware that [libusb-win32](https://sourceforge.net/p/libusb-win32/wiki/Home/) is a separate project and is not maintaned anymore, in fact that discussion is very old (and about a non-HDI device), however I'm guessing also if the idea described there could be feasible ("_The best solution would be if the device wouldn't be visible to Windows anymore after opened by libusb. This could only be achieved by dynamically replacing MS' driver with the libusb driver from your application (requires admin privileges)._").

In the end my questions are:

1. given my particular device (class 0 pedal with two class 3 interfaces), should I use libusb or hidapi?
2. if I have no luck with hidapi, **is there a way with libusb to get exclusive acces to that device under Windows 10? By exclusive access I mean that only my libusb based application should be able to receive inputs from the device.**

Thank you.

diabolo38, 2017-08-29:
Install winusb Back end (with zadig for instance)  for the device and you  will have single  app access to it via libusb. It then to you to handle all hid request  and report handling and ep query

Otacon5555, 2017-08-30:
Thank you very much for your reply. Could you please elaborate a bit more?

I've done what you suggested (installed the WinUSB driver for the device via Zadig), and I noted that now xusb.exe succesfully claims both interfaces (before, claiming the second one failed); however debugging it, while both of them are claimed, other applications are still receiving inputs from the pedal (pressing it is equivalent to press the number 1 key of the keyboard).

Since [`libusb_claim_interface`](http://libusb.sourceforge.net/api-1.0/group__dev.html#ga32fabedf5f13fdecf1cb33acdb19b57a) says "_Interface claiming is used to instruct the underlying operating system that your application wishes to take **ownership** of the interface_", and apparently it is successful, I don't know what I'm missing here.

Thank you.

diabolo38, 2017-08-30:
Humm i would have expected other app not to use any more the device. Its like default USB hid driver is still used. Can you double check in device manager if device  appear under hid/kbd or USB dev. I guess if device is keyboard class type they may be no way to overhide os  but maybe using usbdk?

Otacon5555, 2017-08-31:
"_Its like default USB hid driver is still used_": you are actually right, my bad: only now I've realized that I changed with [Zadig](http://zadig.akeo.ie/) the driver only of one interface out of two! Now, after having changed to WinUSB the driver for both interfaces, other applications do not receive input even when xusb.exe is not running, but that's ok for me.

Now, I guess I could try to use also [libwdi](https://github.com/pbatard/libwdi) to programmatically change the drivers of the usb pedal directly from my libusb application, but that's another story.

I'm new to GitHub, but I guess I can now close this thread...?

Anyway, thank you very much for your help!

closed 2017-08-31 by Otacon5555