libusb/libusb #336: Gamecube Adapter - Possible to set USB Polling?

zDEFz, 2017-09-04:
Hi,

We are using libusb through Zadiq for  Mayflash GameCube Controller Adapter for Wii U  and  Official GameCube Adapter for Wii U.
https://www.smashladder.com/guides/view/26oz/controller-guide-2-0
It would be cool for us to change polling rate to 1000.
Unfornately we can't modify the driver because it needs a certificate.
There is already proof on concept for Linux. https://github.com/HannesMann/gcadapter-oc-kmod

But we need it on Windows as well.
What can we do? 

Many Thanks,

DEF

HannesMann, 2017-09-04:
I've already investigated it on Windows and it can't be done, because Windows only provides a copy function to the `USB_ENDPOINT_DESCRIPTOR` struct so you can't modify the `bInterval` value. 

(https://msdn.microsoft.com/en-us/library/windows/hardware/ff539317(v=vs.85).aspx also explicitly says that "Drivers cannot change it." in regards to the `bInterval` value)

mcuee, 2017-09-10:
>
> This is not a libusb or device driver signature (or Zadig) problem. Will
close this.

closed 2017-09-10 by mcuee