libusb/libusb #339: Problem in read large number of raw data(approx 200000 bytes) from bulk in endpoint.

PratikGajera, 2017-09-12:
Currently,my usb device driver application has been worked on windows platform. But I want to develop it for linux. I wrote driver for linux platform and made successfully communication pipe between driver application and device. Device has used 3 transfer type bulk endpoint. There is bulk out and remaining two are bulk in and problem at second bulk in endpoint (0x82) that when read data from it then only 64 bytes read it insted of 200000 bytes. Two bulk type endpoint are properly working (0x01,0x81). So I don't what I am going wrong. Please find below device information and I used alternet setting 0 under interface.  

**** Device Details ****

    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               1

**** My Application Code ****

int main() {
    libusb_init(); 
     ……………
    libusb_get_device_list(…) /* Get the USB devices list */
     ………………
    libusb_open(…);/*opens your device using VID, PID combination*/
     ……………
    libusb_claim_interface(…) /* claim your device interface */
     ……………
     ……………
     /* my application code */
     ……………
     ……………
     libusb_release_interface(…) /* release your device interface*/
     ……………
     libusb_close(…); /* close your device */
     ……………
     libusb_exit(); /* undo the initialization and do cleanup */
}

mcuee, 2017-10-03:
I think you need to provide more info on your code snippets. Which Linux platform?

mcuee, 2017-11-25:
Also you may want to ask in the libusb-devel mailing list. You may get better response for application related question.

dickens, 2017-12-28:
There are only 3 things that will cause an IN transfer to complete:

1. The requested amount of data has been received.
2. The device sends a short or zero-length packet.
3. The transfer is aborted due to a timeout/disconnect.

The only way you can receive 64 bytes from endpoint 0x82 of the above device is if the device sends a 64-byte packet, which would be considered a short packet since the maximum packet size is 512 bytes.

dickens, 2018-03-25:
No response, so closing this. The OP can reopen if there is additional information to provide.

closed 2018-03-25 by dickens