libusb/libusb #340: Low transfer rate

yecunkui, 2017-09-14:
I am using libusb to develop some performance test routine on Windows 7 carring USB 3.0.
The tested performance is only about 60 MB/s, which is much lower than normal speed of USB 3.0.
What may be the reason?

diabolo38, 2017-09-14:
What is the test debice? 
Reason i see :
The devise is slow.  Max burst not high enough
Fifo are too small device can sustain rate for long enough without delay 
The request used are too small(for bulk you can use 2 MB w/o pb). 
Part of the bandwith may be reserved for iso xfer on other device . 
Some PC BIOS do have option  too limits USB speed.

yecunkui, 2017-09-15:
The device supports USB 3.0, and it's not slow.
The buffer length is set to 0x40000, which is big enough.
Any other reason?

diabolo38, 2017-09-15:
with a single 0x40000 buffer you will not reach top rate maybe 100-200MB/s 
To reach more (300MB/s+) you 'll have to use assync api with multi req or use big sync request 
if your data exchange protocol use short or ZLP packet that is not a problem to use very big request. 
I have no issue streaming out/in data at  250-320MB/s from many device (linux dwc3 based gagdet, fx3...)  under win7 with simple back to back big sync request (nec /ti/intel usb3.0 chipset).

yecunkui, 2017-09-19:
Hi diabolo38, 
Could you please share your code snippet of 'simple back to back big sync request'?

diabolo38, 2017-09-19:
that is more or less what i do  on several device and i reach up 320MB/sec  

```c
int bulk_in_big_req(void *buf, int len,  int time_out, int *len_out){
	int req_len;
	int actual;
	int rc;
	unsigned char *cur;

	assert(len%1024==0); //len shall be a max_packet size multiple assume super-speed 1024 bulk
	cur = (unsigned char*)buf;
	while (len) {
		req_len = MIN(1024*1024*2, len);
		rc = libusb_bulk_transfer( devh, ep_no ,  cur, req_len, &actual, time_out );
		cur  += actual;
		len -= actual;
		if (rc)
			break;
		if (actual!=req_len)
			break; //short ZLP end here
		//todo decrease time out and brake as needed
	}
	*len_out = cur-(unsigned char*)buf;
	return rc;
}
```

yecunkui, 2017-09-20:
Thank you. But I didn't find the key difference between your code and mine. It's very strange.

diabolo38, 2017-09-20:
So your device is slow or use to small xfer payload to reach high rate. I have observed 1 or 2 microframe (125us) between request as host latency and almost the same on device 
On my side this explain low rate for small (<MB)exchange payload. 
Note that Even you can minize latency (assync api on both side) you want ever do more than 8K xfer per second as h/w can't do more than one xfer (round trip exchange) per micro frame

yecunkui, 2017-09-29:
If I enlarge the buffer to 0x400000, there will be an "Exception Code c00000fd" error when running the exe.

yecunkui, 2017-09-30:
I use an USB monitor tool to keep monitoring the USB URB packets. The PC only gets 0x10000 bytes data per bulk transfer, and it sends 1 transfer about every 1 ms, so the transfer rate is limited to 60MB/s. Is there any way to enlarge the data size of each transfer or enlarge the frequency of transfer from PC side?

diabolo38, 2017-10-01:
change your device f/w or optimize it to maximize rate. 
whats sort of h/w is it  ? fx3 ?  linux gadget on some arm soc ?

yecunkui, 2017-10-11:
I have found the reason. It's mainly caused by the application of the device (based on linux), which would affect USB performance greatly in my case.
At the same time the 'length' parameter of libusb_bulk_transfer() on the host (based on Windows 7) will also affect the performance to a certain extent.

mcuee, 2017-10-11:
So close this issue now. Thanks for the updates.

closed 2017-10-11 by mcuee