libusb/libusb #341: Data Loss using libusb-1.0.21 on Ubuntu 16.04 with kernel 4.12.0-041200-generic

spunnadi, 2017-09-22:
There is data loss observed on some occassions(requested data size(s) 17944592 bytes, ) and in some cases I don't get data at all ( requested data size 155093008 bytes ). The usb chip used on the hardware is cypress make CYUSB3014. 

There are no obvious error messages in --enable-debug-log generated outputs of libusb-1.0.21 and neither in attached kernel log with usbfs_snoop enabled
[kern_log.txt](https://github.com/libusb/libusb/files/1323820/kern_log.txt)

Please help as I am on this since a month now and I have no clue as to why this is happening.

mcuee, 2017-10-03:
You may need to provide what you are doing. Code snippets may help.

spunnadi, 2017-10-03:
Apologies for not adding the snippet. Please find more details of my problem:

 In the beginning of the execution of the program a lot of synchronous transfers are successfully executed. A zero length packet follows every synchronous transfer which is always successfully handled(code not shown here as there is no problem in synchronous transfers). 
 
I queue up asynchronous submissions upfront (2 in this case as _submit_async_set_ is set to 2 in the beginning)

`
 for(counter=0;counter<submit_async_set;counter++) 
                {

                        img_transfer[counter] = libusb_alloc_transfer(0);              

                        if(!img_transfer[counter])

                                        printf("\n libusb_alloc_transfer for img_transfer failed!!!\n");
                        // fill information for bulk transfer with calculated receive_bytes which is the expected no. of data bytes. 

                                            libusb_fill_bulk_transfer(img_transfer[counter],devh,0x81,OutputBuf,receive_bytes, Callback_1,0,0);

#ifdef DEBUG
                        puts("UsbTransaction: exposureflag; after libusb_fill_bulk_transfer()");
#endif

                        if((r = libusb_submit_transfer(img_transfer[counter])) < 0) {
                                perror("Submit transfer failed");
                                delete[] OutputBuf; // raw data buffer
                                delete[] sFinalOutputBuf; // processed data buffer
                                delete[] iFinalOutputBuf; // processed data buffer
                                
                        }
                        else
                        {
                                printf("\n Submit #%d Transfer Successfull \n",counter+1);
                        }

                }`

In every callback, I check the size of bytes recieved and do the re-submission as follows: 
`
printf("\n SIZE OF RECEIVED BYTES = %d\n",transfer->actual_length);

                if (transfer->status != LIBUSB_TRANSFER_COMPLETED)
                {
                        printf("img transfer status %d?\n", transfer->status);

                        do_exit = 2;
                        libusb_free_transfer(transfer);
                        transfer = NULL;
                        return;
                }

                else
                {
                        printf("\n Transfer completed !\n");
                        if(((total_set-uiRamps)-nooftimes)>=submit_async_set)
                        {
                         track=track%submit_async_set;

                  
                        if((libusb_submit_transfer(img_transfer[track]))<0) //resubmission of completed transfer
                        {
                                printf("\n Submit transfer failed\n");
                        }
                        else
                        {
                                  printf("\n Submit successfull\n");
                           
                        }
                        }

                }`

Sometimes I do receive expected no. of bytes, sometimes I don't receive any bytes at all and sometimes less no. of bytes. The callback is common for expected intermediate FPGA responses (16 bytes), each of which is followed by zero length packet, as well as for data which runs to few 10's of MB to 100's of MB which is NOT followed by a zero length packet.

mcuee, 2017-11-25:
Probably it is easier to ask in the libusb mailing list. You may get better advice there.

dickens, 2017-12-28:
There are only 3 things that will cause an IN transfer to complete:

1) The requested amount of data has been received.
2) The device sends a short or zero-length packet.
3) The transfer is aborted due to a timeout/disconnect.

If you are not using a timeout on these async transfers, then the only explanation is (2).

dickens, 2018-03-25:
No response, so closing this. The OP can reopen if there is additional information to provide.

closed 2018-03-25 by dickens