libusb/libusb #342: usb_bulk_read/write transfer more faster in win10 than in win 7(V0.9x)

GerryLijsh, 2017-09-25:
I have a program used libusb V0.9x to controlling a usb device（windows）.

In the same cases, the speed of usb_bulk_read/write（）transferring in win 10 is almost
ten times than in win 7.  

Is there any things can be optimized in my codes? Any configure can affect the speed?

Or it is just the diff of platforms.

yecunkui, 2017-09-28:
That's very strange. I use Win7 and the transfer rate is also too low. Still checking ...

diabolo38, 2017-09-29:
small synchronous transfer chunk , usb topologies (h/w controller, hub , devices present on bus) can explain low perf, but given same h/w it shall be very close  win7/8/10. 
Ii personally never noticed such perf issue under win7 in all my app devices and various config.
most often i use  libusbk back end but i wan't expect  much difference with winusb .
also i never use such  old libusb  0.9x  but more  recent one >1.09.

GerryLijsh, 2017-09-29:
Thx，maybe i should mov 0.9x to the latest version.  I really can not find any code reasons to explaining my testing results. I will test 1.x later.

closed 2017-09-29 by GerryLijsh