libusb/libusb #343: Unknown Device

vozan, 2017-09-25:
I connect custom device to pc, on some computer's device manager, I see the device like "unknown device". I see code 43 error on device details. But device is running on other computers. Do you think why I get this error?

diabolo38, 2017-09-25:
May be your device is to slow on powr up or  answering to enumeration and some computer may give up early. I usually see such unknow device when debuging device f/w and braking on enum. You could also have h/w issue what is device ?

vozan, 2017-09-26:
Hi, thank for your answer. It seems hardware problem like your comment. But I have another problem now. I use bulk transfer mode for usb communication.I can send data to/from device and libusb sends back error code,On some computers I get Win32Error about communication.

diabolo38, 2017-09-26:
likely  device  issue,  you should get bulk rd/wr errors only if device stall the ep or for time out.
note that   before to be abble to rd/wr from ep  you have top open device  and then claim interface(s) for all ep to use. Without that do not expect much libusb call to suceed

dickens, 2017-09-27:
This isn't a libusb issue, so closing. Feel free to discuss further on the libusb-devel mailing list.

closed 2017-09-27 by dickens