libusb/libusb #344: issue

dunk8888, 2017-10-03:
Hi can you tell me how i can compile my code,its saying this below,is somthing missing,pkg config? how do i get this? Ime trying to compile a usb project i think its pretty old,ime using windows 7 thanks

gcc -g -Wall -O2 `pkg-config --libs --cflags libusb-1.0`  -c tool.c -o tool.o
/usr/bin/sh: pkg-config: command not found
/usr/bin/sh: gcc: command not found
make: *** [tool.o] Error 127

elfmedy, 2017-10-05:
emm, i think it should be -lusb-1.0 or libusb-1.0.a

LudovicRousseau, 2017-10-05:
I am not sure that using GCC under Windows is a good idea unless you know what you do.
Your problem is not related to libusb but to your build environment.

closed 2017-10-05 by LudovicRousseau

dunk8888, 2017-10-05:
So I need to installbone of those versions.

dunk8888, 2017-10-05:
Oh I see,I didn't,read the second message,is it for use with Linux or some
thing like that.