libusb/libusb #347: LibUsbDotNet Bulk Transfer Usb Communication Win32Error

serhanyilmaz, 2017-10-10:
We have a vendor-specific device for USB connections.

We used LibUsbDotNet Bulk Transfer communication. In the first case, I did 64 byte data transfer. In the second case I made 1024 byte data transfer. In case of 64 byte transfer, there is usually no problem (Problems are rare). In case of 1024 byte transfer, there is a problem (almost always in USB 2.0 port, also rare in USB 3.0 port) and Finally, when doing USB Write, I get Win32Error error in LibUsbDotNet ErrorCode class. 

What causes to this problem?

mcuee, 2017-10-10:
Take note libusbdotnet is a separate project.
https://github.com/LibUsbDotNet/LibUsbDotNet

serhanyilmaz, 2017-10-10:
Thank you so much  for answer to my ask 

Have a nice day.

mcuee, 2017-10-19:
Closed for now. Will reopen if it is linked to libusb.

closed 2017-10-19 by mcuee