libusb/libusb #350: Documentation is not clear on what constitutes an "event" in the asynchronous API

jleben, 2017-10-13:
The documentation says that the `libusb_handle_events` function handles "any pending events". It is not clear what all the possible events are.

This is problematic in the following case: Suppose I call `libusb_submit_transfer` to submit a transfer, and this is the only asynchronous request to libusb from the point of view of the user. Is the completion (or failure) of the transfer the only event that could happen, or are there other events that could happen that do not coincide with the completion of the transfer? In other words, if I then call `libusb_handle_events`, and it returns with success before the expiration of the timeout, is it guaranteed that the transfer is completed or not?

The uncertainty is increased by the fact that the documentation mentions that "waiting for events" involves "poll/select" system calls. Whoever has used those calls directly knows that they can return for many reasons, including simply a signal being raised which is completely unrelated with the file descriptor. If `libusb_handle_events` returns as soon as `poll` or `select` returns, then in the above scenario the transfer might not be completed.

LudovicRousseau, 2017-10-18:
Please provide a Pull Request to fix the documentation.

dickens, 2020-08-18:
I'm going to close this one as I believe the documentation is clear enough to describe how one should determine whether a transfer is done.  Additionally, it is not required (or even helpful) to describe what an "event" means to libusb.  The whole point is that the library abstracts away what it is doing internally and lets the application know when there is something to act on.

To the OP, if you feel so inclined to provide clarification to the documentation, please submit a Pull Request.

closed 2020-08-18 by dickens