libusb/libusb #351: Documentation is unclear about the use of `user_data` field of `struct libusb_transfer`

jleben, 2017-10-13:
The documentation for the `user_data` field reads:
> User context data **to pass** to the callback function

However, functions like `libusb_fill_bulk_transfer` also take `user_data` as a parameter. Moreover, the callback function type `libusb_transfer_cb_fn` does not take a `user_data` as a parameter, but only takes a `libusb_transfer` pointer. This makes me guess that the `user_data` passed to `libusb_fill_bulk_transfer` is copied into the `libusb_transfer` struct which is then passed to the callback.

However, this is not explicitly stated anywhere in the documentation. If this is true, then the documentation for the `user_data` field of `libusb_transfer` should read instead:

> User data pointer that **was passed** to a request such as `libusb_fill_bulk_transfer` and is copied here when the callback is invoked.

Moreover, given the current documentation, a user might manually set the `user_data` field of `libusb_transfer` to the data that they wish to use in a callback, and if my guess is right, the field will then be overridden when the callback is called and the callback will use the data with the wrong assumption.

LudovicRousseau, 2017-10-18:
Please provide a Pull Request to fix the documentation.

patstew, 2017-10-28:
libusb_fill_bulk_transfer doesn't do an awful lot, it's not really a 'request' and you don't need to use it at all. This is the full code:
```
static inline void libusb_fill_bulk_transfer(struct libusb_transfer *transfer,
	libusb_device_handle *dev_handle, unsigned char endpoint,
	unsigned char *buffer, int length, libusb_transfer_cb_fn callback,
	void *user_data, unsigned int timeout)
{
	transfer->dev_handle = dev_handle;
	transfer->endpoint = endpoint;
	transfer->type = LIBUSB_TRANSFER_TYPE_BULK;
	transfer->timeout = timeout;
	transfer->buffer = buffer;
	transfer->length = length;
	transfer->user_data = user_data;
	transfer->callback = callback;
}
```
user_data isn't overwritten by libusb at all. You give a pointer to a transfer to libusb_submit_transfer, and some time later that pointer will be passed to your callback.

commit 666647fa (2020-08-18): Documentation: Improve details of libusb_transfer's 'user_data' field

Closes #351
closed 2020-08-18 by dickens