libusb/libusb #352: Documentation is unclear about thread safety

jleben, 2017-10-13:
The documentation repeatedly states that "libusb is thread safe", without clarifying what exactly is meant by that.

Is calling `libusb_handle_events` with the same parameter from multiple threads thread safe? Based on the documentation, I can guess: *probably* yes (although not certainly).

Is calling `libusb_fill_bulk_transfer` with the same parameters from multiple threads thread safe? Based on my experience and common sense, I would guess: *probably* no (although I can think of an implementation where it is).

Other functions may have various degrees of uncertainty about thread safety.

The claim of thread safety should be applied to individual functions, and the conditions for thread-safety should be clearly specified.

LudovicRousseau, 2017-10-18:
Please provide a Pull Request to fix the documentation.

mcuee, 2017-11-25:
http://libusb.sourceforge.net/api-1.0/caveats.html

Probably the "caveats" page is a good place to talk about the above topic in general.

commit e83522e7 (2020-08-18): Documentation: Add details regarding thread-safety

Closes #352
closed 2020-08-18 by dickens