libusb/libusb #353: Unable to work with device connected to ASMedia 3.1 port

ArnonCh, 2017-10-15:
Hi,

I am facing a similar issue to:
https://github.com/libusb/libusb/issues/227

I'm using windows 8.1 pro, and when I try to detect my device, the function libusb_get_device_list() does not list it.

Inside the function init_device (windows_winusb.c)
When it calls the funcion DeviceIoControl with IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX it fails with the error of no device connected

mcuee, 2017-10-19:
So the first problem seems to be an ASMedia driver problem. Can you see the device with Microsoft tools like USBView?

For the 2nd problem, you may want to provide more details.

ArnonCh, 2017-10-22:
Yes the device is visible with USBView (also shows up in the device manager)

diabolo38, 2017-10-22:
1st issue is certainly on as media (same happen on other drivers) and we can't do anything about it. Unless you need to support win 7 I suggest to use ms native driver and see if we cand dimd a solution for bulk.

ArnonCh, 2017-10-23:
Win7 is not a must for me.
My update above refers to the phenomena when I use ms native driver - every call to libusb_bulk_transfer() on the input ep returns with success, even tho my device is not transmitting anything

mcuee, 2017-10-24:
You will need to provide more details, eg: your code snippets and debug log

photron, 2018-03-06:
I had problems with an ASMedia hub on Windows before and in my case the enumeration changes in libusb 1.0.22-rc3 have fixed the problem now.

mcuee, 2018-03-06:
Thanks for the update. Close this issue for now. To the OP, you may reopen the issue if you still encounter issues.

closed 2018-03-06 by mcuee