libusb/libusb #354: DBUS FD handles not released after close

corby, 2017-10-16:
We have an appliance that uses nearly 100 USB devices at once that get unplugged and plugged back in on a daily basis. 
Everything works great for about 2 days and then I get the "too many open files" error.
When I check lsof I see that my process has over 1 million USB file descriptors open in "(deleted)" mode.

I've checked my code several times and I've made sure I close the USB device when I'm done with it or the device is unplugged (there is no eject button).
I've even gone so far as to:
release_interface
close
unref_device

Just to make sure the reference count reaches zero.

I don't know what else to do other than write my own USB FD handler.
Right now I've added a daily proc restart to systemctl which is a very "Windows" thing to do and bugs me.

Any help would be appreciated.

diabolo38, 2017-10-18:
I have seen similar handle leak in windows (about 3 handle per dev open/close  ) by using ref device and doing similar thing.
After trial I seam leak occur in device list release when a device it contains is still referenced, of course de-referencing the device wan't fix the leak. 

I failed to find the reason and cause inside the lib source in a reasonable short time so i  went for a faster work around.  To avoid leak don't do device ref (claim and open would do anyway)  and only release device list when done with the device (after interface release and dev close) . 

typically what i do 
```
libusb_init()
libusb_get_device_list()
//find device
libusb_open()
libusb_claim_interface()

libusb_release_interface
libusb_close
libusb_free_device_list
libusb_exit
```

doing so  i do not get leak and can do thousand loops  .. safely .

corby, 2017-10-18:
Thanks for the response.
I refactored my code last week and was able to get rid of the FD leak.
I had a separate context for each thread and I think that got me in trouble, I now use the global context and re-ordered the free_transaction, cancel_transaction, and close function to get the ref count to the correct place.

It would be very helpful for debugging if I could somehow get the refcount on a device in a multithreaded app so I can guess where the lock was.

I'll close this and add a feature request for better refcount debugging.

Thanks

closed 2017-10-18 by corby