libusb/libusb #357: Bug: libusb_unref_count on zero ref device crashes

corby, 2017-10-18:
if a device has the refcount set to 0 either by libusb_close or libusb_unref_count, if you call libusb_unref_count again, the library segfaults.
It should return and error code (possibly an int) to indicate that the operation failed.

Justification:
I use golang to integrate with libusb and it works great, but golang is not awesome at handling signals like SIGABRT or SIGSEG.  Returning error codes instead of having to implement signal handlers would be much nicer.

LudovicRousseau, 2017-10-18:
Please provide a Pull Request.

closed 2017-10-18 by corby

dickens, 2017-10-18:
This is not a bug and what you are requesting cannot be done. As soon as the refcount reaches 0, the memory is freed. At that point there is no way to detect an error in a subsequent call.