libusb/libusb #360: How to get USB OTG version on android using libusb?

andymatos, 2017-11-03:
I am using [https://github.com/saki4510t/UVCCamera](url) to capture usb camera preview on android phones. After some tests, I found that high resolutions only works on phones which support usb otg 2.0. On the phones which only support usb otg 1.3, it only works on a very low fps due to the bandwidth limit. If I set resolution and fps to 1280*720@15fps，I only get broken frame from the driver.
So I'd like to know if there is some way to get the usb otg version in libusb and then I can decide the resolution and fps  according to the otg version.

EspoTek, 2017-11-06:
It might be theoretically possible, but libusb wouldn't be the best library to do this.
Most people only use libusb for the actual data transfer itself; for discovering the devices and other things, we use the Android USB API.
Android security is a nightmare to work around if you're not using the official code.

mcuee, 2017-11-25:
I will agree that libusb is not the right tool for this task, in fact as of now, libusb main line code does not really work well under Android due to all the security stuff. There are patches but they are not reviewed and integrated in the main line yet.

closed 2017-11-25 by mcuee