libusb/libusb #363: len is not max_packet size multiple at super-speed 1024 bulk

igok99, 2017-11-12:
Hi all,
from what I understand when using a SS device a data length shall be a max_packet size- multiple of 1024 bulk. But in my case, the data which i am receiving on my host can't be divided by 1024, which leads to a timeout or a big delay. is there a way to handle/avoid this delay when using libusb_bulk_transfer function?

Thanks.

diabolo38, 2017-11-12:
If device transfert  is not max packet multiple then it should not time out it end up in a short packet.  It is actulay the opposite if length is multiple the Device shall emit a zero length to mark transfer end or host must read the exact size

igok99, 2017-11-13:
> If device transfer is not max packet multiple then it should not time out it end up in a short packet-
it can time out when the host asks for 1024 bytes, but actually receives only 800.
Is there a way not to wait for the last 224 bytes?

diabolo38, 2017-11-14:
That is not quite normal they may be some option to enable short and zlp in host side but that is up to me the default in all OS. How sure are you your device sending 800 bytes xfer ?
What kind of end point is it ? Cos for an interrupt ep configured  as 32 byte max packet 800 is an exact packet size multiple.

closed 2017-12-07 by igok99