libusb/libusb #364: libusb-1.0 communicate with printer and then the cups can not print

liuxiaomeng2270, 2017-11-20:
I am developing a application(state monitor) which communicate directly to the printer. The communication function I run to libusb for it provided abunch of functions to read and write with USB.

My system is linux redhat and when I plug in the printer, the system generate a link under /dev/usb/lp0.
When I run my application, which calls 
1.libusb_init();
2.libusb_find_busses();
3.libusb_find_devices();
4.libusb_open_device_with_vid_pid();
5.libusb_free_device_list();
6.libusb_detach_kernel_driver();
7.libusb_claim_interface();
8.use libusb_bulk_transfer() to communicate with printer
9.libusb_release_interface()
10.and then libusb_attach_kernel_driver();
11,libusb_exit();

when the state monitor application exit, and I can not print any thing from CUPS.

the CUPS log shows can not communicate to printer(can not read any thing from printer).

I found when I call libusb_detach_kernel_driver(); the /dev/usb/lp0 link disappears and then I call libusb_attach_kernel_driver() the link /dev/usb/lp0 appears again. but still can not print anything.

when I run echo "somthing" > /dev/usb/lp0 in termail and cat /dev/usb/lp0 , I can get the return string from printer.

For I am not familiar with libusb, I can not figure it out why the printer(CUPS indeed) can print.

Any suggestion will be welcome. 

Thank you.

liuxiaomeng2270, 2017-11-24:
Anybody? Any suggestion?

mcuee, 2017-11-25:
Maybe you can ask CUPS guys to see if they have some suggestions.

It seems libusb is doing alright here.

mcuee, 2017-11-25:
1) /dev/usb/lp0 appears again
2) when I run echo "somthing" > /dev/usb/lp0 in termail and cat /dev/usb/lp0 , I can get the return string from printer.

From the above, this does not seem to be a libusb problem, it could be the printer problem or CUPS problem. I can only guess that the printer has run into some weird state. You can try to add libusb_reset_device() after step 8 or 9 of your program to see if that helps.

dickens, 2017-12-22:
Definitely a CUPS issue. In order to claim the printer class interface, the kernel's built-in usblp driver, which is responsible for creating the lpX device nodes under /dev/usb, must be detached. When done using the interface, the usblp driver can be reattached, which will cause the driver to bind to the interface and create a new lpX device node.

If CUPS can't communicate with the printer after the first instance of the kernel driver is detached, there me some stale information that CUPS is holding on to. There's nothing libusb can do about this.

closed 2017-12-22 by dickens