libusb/libusb #366: libusb_control_transfer hanging issue

chrisgolden, 2017-11-24:
Hi everyone,

I've been working with libusb on Windows, Linux and Mac for some time now without any real issues. However, something happened yesterday which has really stumped me and I can figure out what's going on. I'm using the 'libusb_control_transfer' to upload some data to a camera we created which contains a Cypress FX3. This method has worked without issue on many machines (including lots of customer machines) for at least a year. Then yesterday, it stopped working on one of our Windows 10 production machines. They say that it was working perfectly, then they installed the latest windows 10 update and suddenly it stopped working (This may be a coincidence, and that machine is actually fairly new). 

Anyways, the problem is that the 'libusb_control_transfer' hangs forever even though I have set the time out to 200ms. This doesn't happen every time, it'll send a lot of data before it eventually hangs, but it never gets to the end of the data stream. To rule out an issue on the FX3 itself, I've tried modifying the receiving method to 'do nothing' but the issue still remains.

I managed to download the libusb source code and build it to see if I could get any clues (this was actually surprisingly easy to do, good job!). I've managed to chase the problem down to the 'usbi_poll' method. Usually, when it gets to the 'HasOverlappedIoCompleted' section, the 'poll_fd[_index].overlapped->Interval' value is 0, so it can continue on. However, when it fails, the this value is 259 (STATUS_PENDING). It then heads into the next section, where it gets stuck into the loop of waiting for the handles for a minute, then waking up and repeating. 

I think the real problem might be in the 'winusbx_submit_control_transfer' method, where it calls WinUSBX[sub_api].ControlTransfer(wfd.handle, *setup, transfer->buffer + LIBUSB_CONTROL_SETUP_SIZE, size, NULL, wfd.overlapped). 
I added a little debug code here to see what would happen if I created a loop here which waited until  'wfd.overlapped->Internal' changed to 0. Most of the time, the value gets changed to 0 very quickly (within 10ms). However, when it fails, this value never changes and it get stuck here forever instead. I realise that this is actually just the WinUsb_ControlTransfer method. But there doesn't seem to be much information online about it, particularly when using the Overlapped structure. 

Anyways, if you have any advice on how to proceed here, that would be greatly appreciated. At the moment, I can't really see how to get out of this situation without trying to modify the libusb code (which I'd rather avoid). My main concern is that, if it is something to do with a windows update, we are about to get loads of emails from customers who no longer have a working camera. I'm happy to supply more information / code samples if this helps.

Thanks in advance

Chris

dickens, 2017-12-22:
There is an expectation that when submitting a request using overlapped I/O that both the status and state of the event will be updated when the request completes. Based on your observations, especially your loop that checks the 'Internal' field, it appears that there are situations where either the transfer doesn't complete OR Windows isn't reporting it correctly.

In such cases, are you able to see the request on the device side?

dickens, 2018-03-25:
No response, so closing this. The OP can reopen if there is additional information to provide.

closed 2018-03-25 by dickens