libusb/libusb #367: Intermittent problem loading multiple units via USB, synchronous (Windows 7)

DavidFitz-work, 2017-11-30:
I'm trying to understand why our use of libusb isn't 100% reliable when loading an image to multiple units at the same time.  The load process occurs in many packets, where each packet is a write followed by a read of a response message from the device being programmed.  The problem is intermittent and is seen on a PC read operation.

Here are some more details:
- USB full speed (12Mbps).
- image size about 0.5 MB, sent out in 1024-byte chunks, USB max packet size 64 bytes
- Beagle analyzer shows device's response message is put on the wire, but libusb is getting a read timeout during the error condition (0 bytes reported being read by libusb_bulk_transfer() call that was made during the error)
- setting timeout to 0 does not help
- there are 2 global threads: (opening/closing HW, polling for new attachments/detachments)
- there are 2 threads per connected device:  Tx data and Rx data
- Am using > 12 units at a time, through 2 hubs (one hub is cascaded through another)
- See problems _even with units connected to the 1st hub in the chain_
- Problem does not occur when fewer units are there.  (Problem is related to level of traffic present on the wire)

There is one thread that polls for new devices attached / devices disconnected every 500msec.  If a new device with a particular VID/PID is found, then that device is added.  This function has the following pseudocode (apologies for formatting; indenting is ignored, CR/LFs not applied when inserting code):

if (numDevices >= 0) {
	scan through all numDevices, looking for a specific VID/PID combo
	if(device exists at ppDevices[i]) {
		libusb_get_device_descriptor( )
		if this device has VID/PID we're looking for {
			populate device pointer for new USB device, using following libusb calls for setting some of the elements:
			libusb_get_bus_number( )
			libusb_get_device_address( )
			libusb_get_port_numbers( )
		}
	}
}
libusb_free_device_list( )

I've attached the file if you want to see the above, with indentations intact.
[pseudocode.txt](https://github.com/libusb/libusb/files/1516299/pseudocode.txt)

When I make this thread run less often, the rate of failures drops significantly.  Somehow, the polling is affecting read operations.

Is polling for device attachments/detachments during synchronous operation ok to do?
Do you know if any of the above libusb calls should not be done, or am I missing a call?
Is there a maximum level of traffic that libusb can handle?  PC is i7 core with 12GB RAM.
Looks like a handle isn't being "handled" correctly, but it isn't obvious to me where that would be.

Any help would be appreciated.

diabolo38, 2017-11-30:
I'm not sure the sync api is multi thread safe and meant for that maybe use with onebusb  context per device/could be safer.
I have already face some issue by mixing sync and assync without care on race flag pass to handle event so it may  be somehow what hapen with multi sync call.

You may get more help by asking in the mail list.
 
 About polling for new device to  overcome lake of hotplug  be aware that it can easiky create memory and handle leak you may  be concerned if your app need to run for Long time.

What do you mean By amount of trafic?  Data rate ? Theirs none but the one inherent to your h/w and specific s/w design. They may be some limit about amount of concurrent request runing  at a time and or device open.

mcuee, 2018-04-16:
Close this for now.

closed 2018-04-16 by mcuee