libusb/libusb #369: Failure to cancel ends in infinite loop

jlas, 2017-12-04:
Running 1.0.21 on Linux 64bit

I am seeing infinite looping behavior during tear down of a usb device through a separate library (control looks like this [my app] -> [other lib] -> [libusb]). I am new to this codebase but from a quick investigation it seems that in `sync.c:sync_transfer_wait_for_completion` when an attempt to cancel fails (i.e. `libusb_cancel_transfer` returns non 0) it will continue the event handling indefinitely, and should halt instead.

The traceback within libusb when the infinite loop happens looks like this:

```
poll
handle_events
libusb_handle_events_timeout_completed
libusb_handle_events_completed
sync_transfer_wait_for_completion
do_sync_bulk_transfer
libusb_bulk_transfer
```

And the debug output looks like this:

```
...
[ 9.228022] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[ 9.228028] libusb: debug [usbi_handle_transfer_completion] transfer 0xaddr1 has callback 0xaddr2
[ 9.228034] libusb: debug [sync_transfer_cb] actual_length=16
[ 9.228041] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 9.228044] libusb: debug [libusb_free_transfer] transfer 0xaddr1
[ 9.228049] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms

<application starts tearing down device at this point>

[ 9.240724] libusb: debug [handle_events] poll() returned 1
[ 9.240754] libusb: error [reap_for_handle] reap failed error -1 errno=14
[ 9.240760] libusb: error [handle_events] backend handle_events failed with error -1
[ 9.240766] libusb: error [sync_transfer_wait_for_completion] libusb_handle_events failed: LIBUSB_ERROR_IO, cancelling transfer and retrying
[ 9.240772] libusb: debug [libusb_cancel_transfer] transfer 0xaddr3
[ 9.240779] libusb: debug [discard_urbs] URB not found --> assuming ready to be reaped
[ 9.240784] libusb: debug [libusb_cancel_transfer] cancel transfer failed error -5
[ 9.240789] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 9.240794] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 9.697736] libusb: debug [handle_events] poll() returned 1
[ 9.697752] libusb: debug [handle_events] timerfd triggered
[ 9.697758] libusb: debug [libusb_cancel_transfer] transfer 0xaddr3
[ 9.697764] libusb: warning [handle_timeout] async cancel failed -5 errno=22
[ 9.697770] libusb: debug [disarm_timerfd] 
[ 9.697775] libusb: debug [handle_events] poll() 3 fds with timeout in 0ms
[ 9.697781] libusb: debug [handle_events] poll() returned 0
[ 9.697787] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 9.697792] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[69.757879] libusb: debug [handle_events] poll() returned 0
[69.757902] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[69.757908] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms

... (last 3 lines repeat indefinitely)
```

Just to note, I've looked at other open issues and noticed that #297 has a similar incident where the `libusb_cancel_transfer` encounters an error.

dickens, 2017-12-22:
There's something else going on here that seems likely to be an application bug.

`[ 9.240754] libusb: error [reap_for_handle] reap failed error -1 errno=14`

errno 14 is EFAULT, which means that the kernel encountered an invalid address when attempting to copy data to userspace. The only way this can happen is if the urb allocated for the transfer has been freed prior to being reaped.

What exactly is the application doing when it starts "tearing down the device"?

dickens, 2018-03-25:
No response, so closing this. The OP can reopen if there is additional information to provide.

closed 2018-03-25 by dickens