libusb/libusb #370: libusb_open Fails [Win 10][WinUSB]

gudenau, 2017-12-06:
When attempting to open a Logitech K120 keyboard with libUSB version 1.0.21.11156 on Windows 10 x64 Pro with `libusb_open` it fails with `LIBUSB_ERROR_NOT_SUPPORTED`.

According to Device Manager and Zadig version 2.3 build 701 WinUSB is installed and the keyboard does not respond to inputs.

On October 5th the same piece of code worked without issue to open and read from the keyboard.

[image]
[image]
[image]

dickens, 2017-12-22:
You have installed the WinUSB driver for the interfaces, but there is a parent driver under which these interfaces are bound. Are these interfaces under a "USB Composite Parent" device? If not, libusb will not be able to interact with the device. If it is, please send a debug log so I can see the enumeration process.

If it turns out these interfaces are *not* under a "USB Composite Parent", you should be able to install WinUSB on whatever device that is using Zadig.

gudenau, 2018-01-23:
Sorry about the delay, did not get the message.

How do I check this? This is also the behavior on v1.0.21.11156.

drizzd, 2018-02-15:
It works now. I seems totally random to me why one driver combination works and another does not. There is only one difference in the debug output of xusb up to the call to libusb_open, which is that in the good case there is no mention of interface 1, presumably because the driver name does not match.

I reinstalled the FTDIBUS driver by letting Windows search for a driver automatically on both interfaces. Then in Zadig I replaced the FTDIBUS driver with WinUSB driver only for interface 0.

I did _not_ touch the composite parent driver.

WinBUS on interface 0, WinBUS on interface 1: [xusb-bad.txt](https://github.com/libusb/libusb/files/1727678/xusb-bad.txt)

```
C:\icestick\find_all-example-windows_i386-2>find_all
Number of FTDI devices found: 2
Checking device: 0
ftdi_usb_get_strings failed: -4 (libusb_open() failed)
```

FTDIBUS on interface 0, FTDIBUS on interface 1:

```
C:\icestick\find_all-example-windows_i386-2>find_all
Number of FTDI devices found: 1
Checking device: 0
ftdi_usb_get_strings failed: -4 (libusb_open() failed)
```

WinUSB on interface 0, FTDIBUS on interface 1: [xusb-good.txt](https://github.com/libusb/libusb/files/1727679/xusb-good.txt)

```
C:\icestick\find_all-example-windows_i386-2>find_all
Number of FTDI devices found: 1
Checking device: 0
Manufacturer: Lattice, Description: Lattice FTUSB Interface Cable
```

mcuee, 2018-02-24:
Are you using the latest version of libftdi? 
1) If you just need to use libftdi and not other tools use FTDI vendor driver, then you can use WinUSB to replace the parent driver, it will sure work.
2) If you need to use libftdi on Interface 0 and FTDI vendor driver for Interface 1 (eg: use as a USB serial port), then you just use Zadig to replace  vendor driver to WinUSB only for Interface 0. This is often the use case for many FTDI2232 device.

mcuee, 2018-02-24:
And I will say find_all may not be a very good example.

drizzd, 2018-02-27:
I retrieved the `find_all` binary which I used above from https://github.com/FPGAwars/libftdi-cross-builder/wiki#testing-on-windows-7 ([download](https://github.com/FPGAwars/libftdi-cross-builder/releases/download/v0.2/find_all-example-windows_i386-2.zip)). I also compiled the latest libftdi version, and the behavior was the same.

Please note that I also posted xusb-bad.txt and xusb-good.txt output logs from xusb, which is part of libusb. So I think we do not have to consider libftdi. You are correct that it works if I use libftdi on Interface 0 and the FTDI vendor driver for Interface 1. This corresponds to my good case:

> WinUSB on interface 0, FTDIBUS on interface 1: xusb-good.txt

From what I can tell, working with multiple interfaces is frequently a source of confusion with libusb. In this use case libusb/xusb should be able to operate successfully on interface 0, but its behavior is somehow influenced by the driver installed on interface 1.

I see much room for improvement on libusb side. For example, the error message could indicate that there is an issue only with interface 1, and not with interface 0. There could furthermore be an option to select interface 0, such that interface 1 is ignored from the start. This would enable applications to fix such issues.

mcuee, 2018-03-15:
In reality, this has not much to do with libusb. Rather you have to understand the driver installation. Remember that Zadig or libusbK or libusb-win32 are all not part of libusb project. usbdk is also a seperate project. WinUSB belongs to Microsoft. You may have to deal with the complexity of Windows driver if you want to deal with libusb. But that is not the fault of libusb.

mcuee, 2018-03-15:
You may want to try again with the latest git master on this case: WinBUS on interface 0, WinBUS on interface 1. xusb should work in this case.

mcuee, 2018-04-20:
Close for now. The OP can reopen this if providing more info.

dickens, 2020-08-18:
Going to close this one again.  I don't see any active libusb issue to address.

closed 2020-08-18 by dickens