libusb/libusb #379: How to use libusb for composite device (Microsoft LifeCam)

shbgr, 2018-01-19:
How can I Use libusb for a composite device (Microsoft Lifecam Studio HD for me, USB\VID_045E&PID_0772)?

Whenever I replace the driver for the Composite Parent (with Zadig) the Camera and Microphone devices are lost.

diabolo38, 2018-01-19:
Assigne driver with Zadig only to the interface to manage. If you mess up install once it may be tricky to do a clean driver uninstall and install again. Note that  If the interface you need  is part of the iad for voice or video then it may not be possible to not losses this function

mcuee, 2018-02-24:
Close this issue. If the OP needs further support, please use mailing list for technical support. Thanks.

closed 2018-02-24 by mcuee