libusb/libusb #384: libusb_exit on Window doesn't clear the device list in libusb_context

changyp6, 2018-01-30:
How to reproduce:
1. Plugin a USB device
2. Call libusb_get_device_list
3. Call libusb_exit (libusb_exit reports: libusb: warning [libusb_exit] some libusb_devices were leaked)

If unplug the USB device before calling libusb_exit, everything works normally.

diabolo38, 2018-01-30:
Likely you miss device list release or unref that is causing the leak

changyp6, 2018-01-31:
@diabolo38 I called libusb_device_list_free(). it's the same.

changyp6, 2018-01-31:
Sorry!! It's my mistake.
I forgot to call libusb_unref on the libusb_device object before calling libusb_exit.

After calling libusb_unref on all the libusb_device, libusb_exit won't report errors.
I'll close this issue.

closed 2018-01-31 by changyp6