libusb/libusb #385: libusb_device_list on Windows can only list HID device

changyp6, 2018-01-31:
I'm testing libusb 1.0.22.rc2 on a Windows 7.
When I call libusb_device_list, only HID devices information can be found.
Other devices are missing.

However, the same program with libusb 1.0.21 runs perfectly.

dickens, 2018-01-31:
Could you provide a debug log? Please uncomment  [windows_winusb.c:1187](https://github.com/libusb/libusb/blob/master/libusb/os/windows_winusb.c#L1187) and rebuild the library.

changyp6, 2018-02-01:
@dickens 
Following are the log.
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [0000210c] libusb: debug [libusb_init] created default context
[ 0.004000] [0000210c] libusb: debug [libusb_init] libusb v1.0.22.11297-rc2
[ 0.004000] [0000210c] libusb: debug [windows_init_dlls] Will use CancelIoEx for I/O cancellation
[ 0.005000] [0000210c] libusb: debug [get_windows_version] Windows 7 SP1 64-bit
[ 0.005000] [0000210c] libusb: debug [windows_init_clock] hires timer available (Frequency: 3328173 Hz)
[ 0.005000] [0000210c] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.006000] [0000210c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.008000] [0000210c] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.008000] [0000210c] libusb: debug [winusbx_init] libusbK version: 3.0.6.0
[ 0.009000] [0000210c] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.009000] [0000210c] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.010000] [0000210c] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.012000] [0000210c] libusb: info [windows_init] UsbDk backend is not available
[ 0.012000] [0000210c] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.013000] [0000210c] libusb: debug [libusb_get_device_list]
[ 0.276000] [0000210c] libusb: debug [winusb_get_device_list] #### PROCESSING HUBs {F18A0E88-C30C-11D0-8815-00A0C906BED8}
[ 0.276000] [0000210c] libusb: debug [winusb_get_device_list] PRO: IUSB3\ROOT_HUB30\4&2FBBB235&0
[ 0.276000] [0000210c] libusb: debug [winusb_get_device_list] found new PnP enumerator string 'IUSB3'
[ 0.277000] [0000210c] libusb: debug [winusb_get_device_list] allocating new device for session [1DBC]
[ 0.277000] [0000210c] libusb: debug [winusb_get_device_list] #### PROCESSING DEVs {A5DCBF10-6530-11D2-901F-00C04FB951ED}
[ 0.277000] [0000210c] libusb: debug [winusb_get_device_list] PRO: USB\VID_413C&PID_2107\5&3ACDD96D&0&5
[ 0.277000] [0000210c] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.278000] [0000210c] libusb: debug [get_api_type] matched driver name against HID API
[ 0.278000] [0000210c] libusb: debug [winusb_get_device_list] allocating new device for session [17C4]
[ 0.278000] [0000210c] libusb: debug [winusb_get_device_list] PRO: USB\VID_413C&PID_301A\5&3ACDD96D&0&4
[ 0.278000] [0000210c] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.279000] [0000210c] libusb: debug [get_api_type] matched driver name against HID API
[ 0.279000] [0000210c] libusb: debug [winusb_get_device_list] allocating new device for session [1814]
[ 0.279000] [0000210c] libusb: debug [winusb_get_device_list] #### PROCESSING HCDs {3ABF6F2D-71C4-462A-8A92-1E6861E6AF27}
[ 0.279000] [0000210c] libusb: debug [winusb_get_device_list] PRO: PCI\VEN_8086&DEV_A12F&SUBSYS_06B91028&REV_31\3&11583659&0&A0
[ 0.279000] [0000210c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A12F&SUBSYS_06B91028&REV_31\3&11583659&0&A0' bus number 1
[ 0.279000] [0000210c] libusb: debug [winusb_get_device_list] #### PROCESSING GENs
[ 0.280000] [0000210c] libusb: debug [winusb_get_device_list] PRO: IUSB3\ROOT_HUB30\4&2FBBB235&0
[ 0.281000] [0000210c] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'IUSB3\ROOT_HUB30\4&2FBBB235&0'
[ 0.281000] [0000210c] libusb: debug [winusb_get_device_list] #### PROCESSING HIDs {4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.281000] [0000210c] libusb: debug [winusb_get_device_list] PRO: HID\VID_413C&PID_301A\6&33F51F01&0&0000
[ 0.281000] [0000210c] libusb: debug [winusb_get_device_list] setting HID interface for [1814]:
[ 0.281000] [0000210c] libusb: debug [set_hid_interface] interface[0] = \\.\HID#VID_413C&PID_301A#6&33F51F01&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.281000] [0000210c] libusb: debug [winusb_get_device_list] PRO: HID\VID_413C&PID_2107\6&72C77C1&0&0000
[ 0.281000] [0000210c] libusb: debug [winusb_get_device_list] setting HID interface for [17C4]:
[ 0.281000] [0000210c] libusb: debug [set_hid_interface] interface[0] = \\.\HID#VID_413C&PID_2107#6&72C77C1&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.281000] [0000210c] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.282000] [0000210c] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.282000] [0000210c] libusb: debug [libusb_get_device_descriptor]
[ 0.282000] [0000210c] libusb: debug [libusb_unref_device] destroy device 1.0

[40.613408] [0000210c] libusb: debug [libusb_get_device_list]
[40.943507] [0000210c] libusb: debug [winusb_get_device_list] #### PROCESSING HUBs {F18A0E88-C30C-11D0-8815-00A0C906BED8}
[40.943507] [0000210c] libusb: debug [winusb_get_device_list] PRO: IUSB3\ROOT_HUB30\4&2FBBB235&0
[40.943507] [0000210c] libusb: debug [winusb_get_device_list] found new PnP enumerator string 'IUSB3'
[40.947508] [0000210c] libusb: debug [winusb_get_device_list] allocating new device for session [1DBC]
[40.951510] [0000210c] libusb: debug [winusb_get_device_list] #### PROCESSING DEVs {A5DCBF10-6530-11D2-901F-00C04FB951ED}
[40.957511] [0000210c] libusb: debug [winusb_get_device_list] PRO: USB\VID_4255&PID_000E\123456789ABC
[40.962513] [0000210c] libusb: debug [get_api_type] driver(s): libusb0
[40.965514] [0000210c] libusb: debug [get_api_type] matched driver name against libusb0
[40.969515] [0000210c] libusb: debug [winusb_get_device_list] allocating new device for session [1EC0]
[... cut ...]
[41.061543] [0000210c] libusb: debug [set_hid_interface] interface[0] = \\.\HID#VID_413C&PID_2107#6&72C77C1&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[41.061543] [0000210c] libusb: debug [libusb_unref_device] destroy device 0.0
[41.065544] [0000210c] libusb: debug [libusb_unref_device] destroy device 0.0
[41.069545] [0000210c] libusb: debug [libusb_unref_device] destroy device 0.0
[41.074546] [0000210c] libusb: debug [libusb_get_device_descriptor]
[41.078548] [0000210c] libusb: debug [libusb_unref_device] destroy device 1.0

changyp6, 2018-02-01:
@dickens 

I seem to have found the root cause of this issue.
The device VID_4255 is found in PASS_DEV stage, however, VID_4255 is NOT found in PASS_GEN stage, but all the device is initialized in PASS_GEN stage, that's why VID_4255 is NOT listed.

Hope this information is useful.

changyp6, 2018-02-01:
@dickens 

Another update about this issue.
My USB device is listed under USB\VID_xxxx&PID_xxxx in my libusbK driver INF file.

My Windows system only has USB3.0, which means there are only IUSB3 root hub.
In this case, libusb 1.0.22 cannot list my device in USB node, but libusb 1.0.21 can list them.

dickens, 2018-02-02:
Thank you for providing the log. I know what the problem is and will provide a fix tonight.

diabolo38, 2018-02-14:
I'm wondering if i'm facing ? 

My device (usb 2.0 FS + libusbk back-end driver installed) is missing  in dev list with 1.0.22rc2 
this show on  win7-32 Virtual  Box  (same test , same VM work with dll 1.0.21)

1.022rc2 also work on other  setup 
- win7-64 VBox host machine 
- win8.1-32 VBox 
- Win10 VBox 

Debug log  Win7-32 WBox libusb1.0.22rc2 dll with enum_debug  uncomented
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [00000778] libusb: debug [libusb_get_device_list]
[ 0.062500] [00000778] libusb: debug [winusb_get_device_list] #### PROCESSING HUBs {F18A0E88-C30C-11D0-8815-00A0C906BED8}
[ 0.078125] [00000778] libusb: debug [winusb_get_device_list] PRO: IUSB3\ROOT_HUB30\4&24054718&0
[ 0.078125] [00000778] libusb: debug [winusb_get_device_list] found new PnP enumerator string 'IUSB3'
[ 0.078125] [00000778] libusb: debug [winusb_get_device_list] allocating new device for session [1548]
[ 0.078125] [00000778] libusb: debug [winusb_get_device_list] #### PROCESSING DEVs {A5DCBF10-6530-11D2-901F-00C04FB951ED}
[ 0.078125] [00000778] libusb: debug [winusb_get_device_list] PRO: USB\VID_0553&PID_0608\0.1-5-G6F02-DIRTY
[ 0.078125] [00000778] libusb: debug [get_api_type] driver(s): libusbK
[ 0.078125] [00000778] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.078125] [00000778] libusb: debug [winusb_get_device_list] allocating new device for session [EAC]
[ 0.093750] [00000778] libusb: debug [winusb_get_device_list] #### PROCESSING HCDs {3ABF6F2D-71C4-462A-8A92-1E6861E6AF27}
[ 0.093750] [00000778] libusb: debug [winusb_get_device_list] PRO: PCI\VEN_8086&DEV_1E31&SUBSYS_00000000&REV_00\3&267A616A&0&60
[ 0.093750] [00000778] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_1E31&SUBSYS_00000000&REV_00\3&267A616A&0&60' bus number 1
[ 0.093750] [00000778] libusb: debug [winusb_get_device_list] #### PROCESSING GENs
[ 0.093750] [00000778] libusb: debug [winusb_get_device_list] PRO: IUSB3\ROOT_HUB30\4&24054718&0
[ 0.093750] [00000778] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'IUSB3\ROOT_HUB30\4&24054718&0'
[ 0.093750] [00000778] libusb: debug [winusb_get_device_list] #### PROCESSING HIDs {4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.109375] [00000778] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.109375] [00000778] libusb: debug [libusb_get_device_descriptor]
usb enum l62 dev  0 8086_1E31 bus 1 addr 0
[ 0.109375] [00000778] libusb: debug [libusb_unref_device] destroy device 1.0
[D] ASPIC_COMMS_Enum_Devices return status=0 , 0 devices

Win7-32 VM libusb-1.0.21 dll (no enum debug)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [00000ffc] libusb: debug [libusb_get_device_list]
[ 0.015625] [00000ffc] libusb: debug [windows_get_device_list] allocating new device for session [2FA]
[ 0.015625] [00000ffc] libusb: debug [get_api_type] driver(s): iusb3hub
[ 0.015625] [00000ffc] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015625] [00000ffc] libusb: debug [windows_get_device_list] allocating new device for session [1A]
[ 0.015625] [00000ffc] libusb: debug [windows_get_device_list] extra GUID: {6C4A1444-43A7-E9F9-CE31-4409B00D6E8E}
[ 0.015625] [00000ffc] libusb: debug [windows_get_device_list] allocating new device for session [8B]
[ 0.031250] [00000ffc] libusb: debug [init_device] got bus number from ancestor #2
[ 0.031250] [00000ffc] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.031250] [00000ffc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 67 bytes)
[ 0.031250] [00000ffc] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 1): '\\.\USB#VID_0553&PID_0608#0.1-5-G6F02-DIRTY'
[ 0.031250] [00000ffc] libusb: debug [windows_get_device_list] found existing device for session [1A] (1.0)
[ 0.031250] [00000ffc] libusb: debug [init_device] (bus: 1, addr: 1, depth: 0, port: 0): '\\.\IUSB3#ROOT_HUB30#4&24054718&0'
[ 0.031250] [00000ffc] libusb: debug [get_api_type] driver(s): libusbK
[ 0.031250] [00000ffc] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.031250] [00000ffc] libusb: debug [windows_get_device_list] found existing device for session [8B] (1.2)
[ 0.046875] [00000ffc] libusb: debug [get_api_type] driver(s): libusbK
[ 0.046875] [00000ffc] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.046875] [00000ffc] libusb: debug [libusb_get_device_descriptor]
usb enum l62 dev  0 0553_0608 bus 1 addr 2
[ 0.046875] [00000ffc] libusb: debug [libusb_get_device_descriptor]
usb enum l62 dev  1 8086_1E31 bus 1 addr 1
[D] ASPIC_COMMS_Enum_Devices return status=0 , 1 devices
PASS device detect
1 device present

closed 2018-02-24 by dickens

diabolo38, 2018-02-26:
I confirm it fix issue under vwin7-Virtual Box  with only  USB3 thank

aksonlyaks, 2019-01-26:
Dear all,

If this issue is resolved? I have faced similar issues in two of my laptop alongwith USB 2.0, a USB 3.0 port is also present. 

Thanks

commit cd7aeec8 (2018-02-24): Windows: Always search for "USB" PnP enumerator class

On systems running Windows 7 and earlier that only have a USB 3.0 host
controller, devices may not be listed correctly if the root hub PnP
enumerator is anything other than "USB". This regression was introduced
in the recent enumeration refactoring (commit 71a779d07).

Closes #385