libusb/libusb #389: libusb_open_device_with_vid_pid fails [Win10] [WinUSB]

tgotic, 2018-02-04:
When attempting to open a WinChiphead CH341A device with latest libUSB 1.0.22 (WinUSB backend) on Windows 10 x64, libusb_open_device_with_vid_pid fails:
```
libusb: warning [winusb_get_device_list] could not retrieve port number for device 'USB\VID_1A86&PID_5512\7&35768161&0&2': [13] The data is invalid.
libusb: warning [init_device] could not get node connection information for device 'USB\VID_1A86&PID_5512\7&35768161&0&2': [87] The parameter is incorrect.
```
The same error is with libusb 1.0.21.

Opening the device with UsbDk backend is successful.

mcuee, 2018-02-24:
Please help to provide the full debug log. Thanks. How do you install the WinUSB driver?

tgotic, 2018-02-24:
Please find attached [ch341a.log](https://github.com/libusb/libusb/files/1754340/ch341a.log) libusb debug log. I also attached report from "USB Device Tree Viewer" [ch341a_usbtreeview.txt](https://github.com/libusb/libusb/files/1754341/ch341a_usbtreeview.txt).
This device is not using WinUSB driver. I figured that later, sorry.
I was focused on error
`[winusb_get_device_list] could not retrieve port number for device 'USB\VID_1A86&PID_5512\7&35768161&0&2': [13] The data is invalid.`
and tried to find out why
`pSetupDiGetDeviceRegistryPropertyA(*dev_info, &dev_info_data, SPDRP_ADDRESS, NULL, (PBYTE)&port_nr, sizeof(port_nr), &size)`
is failing on this device. At first, there was registry value for Address at
`HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Enum\USB\VID_1A86&PID_5512\7&35768161&0&2 `. When I uninstalled device and drivers and installed them again, Address didn't show up. Then I modified code to fetch port from registry value
LocationInformation with SPDRP_LOCATION_INFORMATION and finally got message that libusb is not working with CH341A device driver.
Code modification in function `winusb_get_device_list`:
```
if (!pSetupDiGetDeviceRegistryPropertyA(*dev_info, &dev_info_data, SPDRP_ADDRESS,
      NULL, (PBYTE)&port_nr, sizeof(port_nr), &size) || (size != sizeof(port_nr))) {
  
  char location[128] = { 0 };
  size = 0;
  usbi_warn(ctx, "could not retrieve SPDRP_ADDRESS for device '%s': %s", dev_id, windows_error_str(0));

  if (pSetupDiGetDeviceRegistryPropertyA(*dev_info, &dev_info_data, SPDRP_LOCATION_INFORMATION,
        NULL, (PBYTE)location, sizeof(location), &size)) {
    
    size = sscanf(location, "Port_#%04u", &port_nr);
  }
  else
    usbi_warn(ctx, "could not retrieve SPDRP_LOCATION_INFORMATION for device '%s': %s", dev_id, windows_error_str(0));
  }
  usbi_dbg("port_nr=%02X size=%lu", port_nr, size);
```

tgotic, 2018-02-24:
I managed to install CH341A with WinUSB drivers according to [Installing WinUSB by specifying the system-provided device class](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb-installation). Now device is working properly even without UsbDk backend. See [ch341_winusb.log](https://github.com/libusb/libusb/files/1754689/ch341_winusb.log)

closed 2018-02-24 by tgotic