libusb/libusb #390: libusb 1.0.22 fails on device in low power state

tgotic, 2018-02-04:
The latest libusb 1.0.22 fails on libusb_set_configuration() and libusb_claim_interface() if device is in low power state:
`libusb: warning [windows_assign_endpoints] could not read config descriptor: error -5`

libusb 1.0.21 doesn't return error and device is successfuly claimed.

tgotic, 2018-02-24:
Please find attached [pickit.log](https://github.com/libusb/libusb/files/1754346/pickit.log) libusb debug log. I also attached report from "USB Device Tree Viewer" 
[pickit_usbtreeview.txt](https://github.com/libusb/libusb/files/1754348/pickit_usbtreeview.txt).
Error happens when Microchip Pickit2 device is attached to USB and idle for some time.
It goes to D3 state and second config descriptor cannot be fetched.
In `cache_config_descriptors` function, second call to 
`DeviceIoControl(hub_handle, IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION, &cd_buf_short, size, &cd_buf_short, size, &ret_size, NULL))` 
fails, code breaks from `for` loop with `LIBUSB_ERROR_IO`, and both config_descriptors are freed. Because of that, later in `windows_assign_endpoints` there is no config descriptor and function fails with error `LIBUSB_ERROR_NOT_FOUND (-5)`.

Now, is it necessary to discard all valid config descriptors if one cannot be fetched, especially in dummy call?
If I change code in `cache_config_descriptors` that `for` loop breaks with `LIBUSB_SUCCESS`:
```
if (!DeviceIoControl(hub_handle, IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION, &cd_buf_short, size,
  &cd_buf_short, size, &ret_size, NULL)) {
  usbi_info(ctx, "could not access configuration descriptor (dummy) for '%s': %s", priv->dev_id, windows_error_str(0));
  LOOP_BREAK(LIBUSB_SUCCESS);
}

if ((ret_size != size) || (cd_buf_short.desc.wTotalLength < sizeof(USB_CONFIGURATION_DESCRIPTOR))) {
  usbi_info(ctx, "unexpected configuration descriptor size (dummy) for '%s'", priv->dev_id);
  LOOP_BREAK(LIBUSB_SUCCESS);
}
```
valid config descriptor is not destroyed, and device can be accessed. Please check [pickit2.log](https://github.com/libusb/libusb/files/1754350/pickit2.log) libusb debug log from modified code.

mcuee, 2018-03-15:
Can you try the latest git master?

mcuee, 2018-03-15:
BTW, it is recommended to use HIDAPI for HID device for better cross-platform support.

tgotic, 2018-03-15:
There is still LIBUSB_ERROR_NOT_FOUND and `[usbi_sanitize_device] zero configurations, maybe an unauthorized device` as in prevous pickit.log.
`memset(&conn_info, 0, sizeof(conn_info))` doesn't solve the problem.

commit 2da989c2 (2018-03-17): Windows: WinUSB: Allow caching config descriptors to fail

Certain buggy devices may not provide all the advertised configuration
descriptors. Prior to this commit, failure to cache any one of the
descriptors would result in all of the descriptors being freed and the
device being inaccessible. Work around this by continuing on if fetching
a configuration descriptor fails. The device may still be usable if the
descriptor for the current configuration was successfully retrieved.

Closes #390
closed 2018-03-17 by dickens