libusb/libusb #391: power per port control in windows and uhubctl not working

Manel3, 2018-02-05:
Dear Sirs,

I would like to control power per port in windows as I can do in Linux (tested).
I have compiled uhubctl for windows and tried with usblib 1.0.21 and compiled also 1.0.22rc2 and activating debug I see there is with both an error on libusb_open the usb hub device:

[ 0.187506] [000002d8] libusb: debug [libusb_open] open 1.0
[ 0.187506] [000002d8] libusb: debug [winusb_open] unsupported API call for 'open' (unrecognized device driver)
[ 0.187506] [000002d8] libusb: debug [libusb_open] open 1.0 returns -12

Default driver of the USB HUB (D-Link DUB-H4 H/W Ver.: D1) is USBHUB3.SYS, I have tried also to replace it using zadig-2.3 by libusbK, libusb-win32, and WinUSB having same error.

Any chance to have this working?
Is it not possible to read/write directly the hardware?

Best regards,
Manel

mvp, 2018-02-13:
@Manel3, -12 is `LIBUSB_ERROR_NOT_SUPPORTED`. Perhaps you could show where exactly this open call fails? Can you find exact line in source code (`uhubctl` I suppose?) that is returning this error? As far as I can tell, `uhubctl` should not crash on Windows except for this control transfer call that is turning power off/on:

https://github.com/mvp/uhubctl/blob/master/uhubctl.c#L635-L639

This is because of libusb using `winusb.sys` driver, which according to Microsoft does not support [necessary USB control requests](https://social.msdn.microsoft.com/Forums/sqlserver/en-US/f680b63f-ca4f-4e52-baa9-9e64f8eee101).

Also, perhaps we should move this issue from libusb to uhubctl.

Manel3, 2018-02-14:
Hi,
I have tried to use libusb compiled with usb**d**k support.
usbdk allows redirect USB devices and I Iiked the idea of having full control of the USB device, that could do the job.

Unfortunately that does not work, I have open an issue:
https://github.com/daynix/UsbDk/issues/55
It looks like hubs can not be redirected, only USB devices.

Not sure that is a uhubctl issue, but I am going to keep them informed.

Any idea on how to walk arround this problem?

mvp, 2018-02-14:
@Manel3, uhubctl is strictly using standard libusb API to control power. However, key interface to make it work is to send control message directly to the hub itself. As far as I know, `winusb.sys` doesn't allow to open USB hub as a device to send control messages (unlike Linux, Mac or FreeBSD), and that's what prohibits uhubctl from working.

If usbdk is actually using winusb.sys as low level interface to hardware, it's not going to work. But, if usbdk implements full USB stack and doesn't need winusb.sys services altogether, it has chance to work. I don't know much, if anything, about usbdk internal design, but I suspect that it doesn't implement full stack.

mcuee, 2018-02-24:
I do not think usbdk uses winusb as the low level interface. But I think usb HUB is not intended to be supported by usbdk. So you are out of luck here.

mcuee, 2018-02-24:
Since usbdk side has closed your issue, I will close the issue here as well.

mvp, 2018-02-24:
@mcuee, I actually think that is issue rightfully belongs to libusb. After all, it is libusb fault that it cannot send standard control message to the USB hub on Windows - it can do so on Linux, Mac, FreeBSD and other posix platforms.

mcuee, 2018-02-24:
I mean we hit the limitation of libusb Windows. I will change this to won't fix unless someone steps in to develop that backend.

Manel3, 2018-02-24:
I am going to try to create a KMDF device lower filter to at least set and clear power port feature. Not sure I will success, that would be my first driver.

Now still researching and documenting: https://github.com/manel3/usbpower

mvp, 2018-02-24:
Awesome, @Manel3! I hope you can find a solution for this...

mcuee, 2021-04-07:
Updated the Wiki and then I will close this issue.
https://github.com/libusb/libusb/wiki/Windows#Known_Restrictions

closed 2021-04-07 by mcuee