libusb/libusb #393: changing the usb data wire signal

mael15, 2018-02-12:
hi everyone!
I want to use an USB activated Switch (Relay) like for Arduino, for my Windows PC. The circuit board has three connection pins for the USB-wire one for 5Volt one for Ground and one for Data. When I switch on my computer the USB-Relay switches immediately on. I am now looking for a software solution to get control over the switch by changing the signal from the USB data wire. 
there is no special driver, i have to manipulate the usb port itself. the example that lists the usb devices works fine, but now what?! i use c++ for my windows 32bit program.
Can somebody help me?

slyshykO, 2018-02-12:
USB interface should have at least 4 wires. If less there is not USB.

mael15, 2018-02-12:
hanks, the data plus wire is not connected.

slyshykO, 2018-02-12:
USB use differential signaling for transferring data. It is mean that data+ and data- used together. There is not possible use them separately.

mael15, 2018-02-13:
i would achieve the desired effect if i could somehow deactivate the whole usb port altogether. is it possible?

mvp, 2018-02-13:
You can deactivate USB port using only software, but only for hubs that support it. Check this out: https://github.com/mvp/uhubctl
For Windows, you might be able to use [`devcon`](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/devcon) to disable/enable USB port.

mael15, 2018-02-14:
thank you @mvp! uhubctl does not work on windows: 

> While uhubctl compiles on Windows, USB power switching does not work on Windows because libusb is using winusb.sys driver, which according to Microsoft does not support necessary USB control requests. This may be fixed if libusb starts supporting different driver on Windows. 

I will try devcon.

mael15, 2018-02-14:
there is an additional problem: i have the relay on a very simple board that does not give me a vendor id or product id:
[image]
[image]
is there a way to deactivate one specific port regardless of the connected device? i could always use the exact same port on the exact same mainboard.

patstew, 2018-02-15:
That device is not USB at all. It's supposed to be controlled by a GPIO.

alex9099, 2018-02-16:
It *isn't* possible to do gpio controlling trough usb, maybe you can power on and off the usb port as @mvp saied, but driving a relay trough usb might not be the best idea.

If you have an arduino you can create a serial connection and control the relay trough the arduino

pbatard, 2018-02-21:
This is clearly not an issue that can be addressed by libusb (which has no control whatsoever over the individual data+ and data- USB signal lines) &rarr; Closing.

closed 2018-02-21 by pbatard