libusb/libusb #394: wIndex always set to 0 with libusb_control_transfer()

kuro68k, 2018-02-21:
I'm trying to debug an issue with dfu-util and I think it's a problem with libusb.

dfu-util calls libusb_control_transfer() with wIndex set to 1, but my device (a custom USB device) receives the packet with wIndex set to 0. The rest of the packet is fine, it's just that field that is wrong.

I verified with Network Monitor packet capture and a logic analyzer that the sent byte is in fact zero.

```
libusb_control_transfer(
	0x005b7800,		// handle
	0xA1,			// bRequestType
	DFU_GETSTATUS,		// bRequest (0x03)
	0,			// wValue
	1,			// wIndex
	buffer,			// data*
	6,			// wLength
	5000);			// timeout
```

The captured packet is:

```
   - ControlTransfer: 
    + Urb: Function = URB_FUNCTION_CONTROL_TRANSFER, Length = 6, Flags 0xb
    - SetupPacket: 
     + bmRequestType: (Class request) 0xa1
       bRequest: 0x3
       wValue: 0 (0x0)
       wIndex: 0 (0x0)
       wLength: 6 (0x6)

A1 03 00 00 00 00 06 00 
```

I was only able to test on Windows, I don't have a suitable Linux box at the moment. wIndex is usually used as the interface number for control transfers, and it's unusual to use values other than zero but in this case it's necessary for a runtime DFU interface.

pbatard, 2018-02-21:
You are most likely running into the WinUSB limitation that is documented in the [Know Restrictions](https://github.com/libusb/libusb/wiki/Windows#Known_Restrictions) for Windows on the Wiki:

> With WinUSB, when `LIBUSB_RECIPIENT_INTERFACE` is used for the transfer, the WinUSB DLL forces the low byte of wIndex to the interface number, regardless of what you set it to.

You will have to use a different driver than WinUSB. However, be mindful that libusb can only go as far as the underlying driver allows it to, and there are restrictions associated with the use of each driver, so it's always a good idea to check the documentation before opening an issue.

As I believe your question should have been answered, I will close this request. Feel free to comment further if you think this is not the case.

kuro68k, 2018-02-21:
Ah yes, that's most likely it. Thanks, I will look at work-arounds.

kuro68k, 2018-02-22:
Having checked the code there is a call to libusb_claim_interface() with the interface number set to 1. That call succeeds. However, WinUSB is still setting the wIndex to 0. Looking at the libusb source, it doesn't actually call any WinUSB from libusb_claim_interface(), it's purely internal libusb stuff.

So if you wouldn't mind offering some assistance here, I'm thinking that there should be a call to WinUsb_GetAssociatedInterface(). I think that's abstracted through windows_usb_api_backend->claim_interface.

Does that sound right?

kuro68k, 2018-02-22:
I'm not sure this is the restriction you mentioned. Here is a debug trace:

```
Claiming USB DFU Interface...
[ 0.188010] [00001288] libusb: debug [libusb_claim_interface] interface 1
[ 0.188010] [00001288] libusb: warning [winusbx_claim_interface] auto-claimed interface 0 (required to claim 1 with WinUSB)
[ 0.188010] [00001288] libusb: debug [winusbx_claim_interface] claimed interface 1
[ 0.188010] [00001288] libusb: debug [windows_assign_endpoints] no endpoints found for interface 1
Setting Alternate Setting #0 ...
[ 0.191010] [00001288] libusb: debug [libusb_set_interface_alt_setting] interface 1 altsetting 0
[ 0.191010] [00001288] libusb: debug [windows_assign_endpoints] no endpoints found for interface 1
Determining device status: [ 0.193011] [00001288] libusb: debug [libusb_alloc_transfer] transfer 0003205C
[ 0.193011] [00001288] libusb: debug [libusb_submit_transfer] transfer 0003205C
[ 0.195011] [00001288] libusb: debug [winusbx_submit_control_transfer] will use interface 0
```

You can see that it does in fact claim interface 1 successfully, but the fails to find any endpoints on it (DFU doesn't use endpoints, only control transfers). After that it uses interface 0 instead of interface 1.

Just before the last debug message we have the following code:

```
	current_interface = get_valid_interface(transfer->dev_handle, USB_API_WINUSBX);
	if (current_interface < 0) {
		if (auto_claim(transfer, &current_interface, USB_API_WINUSBX) != LIBUSB_SUCCESS)
			return LIBUSB_ERROR_NOT_FOUND;
	}
```

The first line sets `current_interface` to 0 and the if() evaluates to false, so the control transfer gets sent to interface 0. `get_valid_interface` returns the FIRST valid interface for control transfers, not the currently claimed one...

Is this a bug? If I use a break point and then manually force current_interface to 1 the transfer control works perfectly and the DFU interface responds as expected.

pbatard, 2018-02-22:
In that case, you will have to propose a patch for `get_valid_interface()` that "remembers" whether an interface was claimed or auto-claimed (we have to auto-claim interface 0 with WinUSB to be able to claim interface 1), and that avoids returning auto-claimed interfaces if an interface was manually claimed.

For the record, I am __no longer__ involved with updating the libusb code (I was only trying to point out to a documented restriction), and, as you should be able to judge from the number of github issues that get unresolved, libusb is too severely understaffed at the moment to expect someone to patch something for you, unless you do it yourself. Hopefully, you can contribute to help improve the library, as, realistically, waiting for somebody else to do that work for you is unlikely to lead to anything at this stage...

kuro68k, 2018-02-22:
I am proposing to patch it myself, yes. I was just after some advice on the best way to fix this issue, which you have kindly provided.

One option is to cache the last interface selected by libusb_claim_interface(). I'm just worried it will break other stuff... It looks like this behaviour is intentional, as in it's not a bug in the Windows implementation or anything like that. So maybe the best solution is to add a new function and also patch dfu-util to use it.

As I am planning to add some other features to dfu-util I might try that. Thanks for your support and hard work thus far, pbatard.

pbatard, 2018-02-22:
Great. Since you are working on a possible patch, I will reopen this issue.

kuro68k, 2018-02-27:
Fix: https://github.com/libusb/libusb/pull/398

mcuee, 2020-08-31:
Close this for now since #398 is closed.

closed 2020-08-31 by mcuee

Elmue, 2025-08-09:
Although this is closed since many years, lots of people still come here from Google and read the wrong information here which I will correct now.

> You are most likely running into the WinUSB limitation that is documented in the [Know Restrictions]
> (https://github.com/libusb/libusb/wiki/Windows#Known_Restrictions) for Windows on the Wiki:
>
> With [...]

This is wrong becasue the libusb Wiki is wrong.
One would suppose that people working on a libusb library should be USB experts, but they are not.
And others like the people in this thread are copying the wrong information without questioning it.

You do NOT need another driver.
You can do this with WinUSB if you know how to do it correctly.

I wrote a detailed correction to this Wiki.
Please read it here:
https://github.com/libusb/libusb/issues/1675

kuro68k, 2025-08-09:
I suppose the issue is that libusb needs a fairly hefty patch to fix this, because it's based on the Linux model.

Have you got a fix for your project?

Elmue, 2025-08-12:
> WCID for whole device ( https://github.com/pbatard/libwdi/wiki/WCID-Devices )

This is a very good and very well explained Wiki.
But I was shocked to find the same wrong information there:

> IMPORTANT NOTE 1: There is a bug/limitation in WinUSB that will force the wIndex 
> of any Interface Request to the interface number. This means that, if you are using WinUSB
> to verify the content of your Extended Properties Feature Descriptor,
> you won't be able to retrieve it (unless it is only defined for interface #5).

The information that this is a bug is spread all over the internet.
I'am very surprised that I'am the first to find that this is not a bug although I'am programming WinUSB for the first time while others are working with it since years. One spreads a misinformation and all the others copy it without questioning it.

> Funny and strange anyway that google end up poiting to libusb for winusb microsoft lack of documentation/example

This is because I searched specificly for "WinUsb_ControlTransfer sets wIndex to zero"
Obviously it is shame that Microsoft gives us such a sparse documentation for this command.
But as a Windows programmer all my life I'am used to this.
I find more information on StackOverflow than on Microsoft.com.

But I have to say that the vast majority of projects on Github has no documentation AT ALL.
And also many software on Linux has worse documentation than what we get from Microsoft.

Generally software developers like programming but they hate to invest one minute in writing a documentation.
There are many many people on Github who dump their buggy code which they never finished and never wrote a documentation and think that others will fix their bugs and try to understand their code which does not have even one comment, not any error handling. At least from Microsoft we get much better quality.