libusb/libusb #399: host (with linux ubuntu OS)  keep old Toggle after clear halt to usb device

reemba, 2018-02-28:
After sending the device (FS device) clear halt (and clear halt sent well) when trying to write to the usb device the "out" sent with toggle 1 and not 0 (like it should be after clear halt).
OS: Linux ubuntu

usb recording on attached file
[image]

mcuee, 2018-03-04:
Which version of ubuntu are you using? I think this has nothing to do with libusb. Most likely the device does not respond well.

reemba, 2018-03-05:
I'm using Ubuntu 16.
no,usb device responds well (i have full usb trace record).
it may be OS problem? (host need to send "OUT" token with toggle 0 but doesn't clear the Toggle)

diabolo38, 2018-03-06:
Some tough it that can help you ..

Have you try different master /controller or  root hub port , add one layer hub ...see if it change anything ?  Is theirs any hub between your device and your analyzer ? 

You said device is FS but on your analyzer trace "clr halt" xfer 161 is HS ?
also  xfer 72 is a split out to your device endpoint ? that usualy indicate an hub is acting arround and this could be the problem.  

Also take care of usb analyzer view "xfer"  mode this is easy to understand but be quiet far away the timing reality . It usualy wan't show concurrency real time execution of all xfer on bus it may  be xfer completion time order not start timer order. So if you don't pay atention at transastion start time and or full detail then  it's  far from the bus time  reality . The low level (non xfer) trace view is less readable but show the reality timing and could clearly expose that  your control clr_ht span over the "wr"  even that's not what the xfer view show at first .
(This is one reason i prefer using beagle to  lecroy analyzer)

Are  you not trying to do the write  ep-2 out prior concurrent to the ep-2 clr-halt ?
I suspect doing this is kind of incorrect and could give unpredictable behavior. (you can change the toglle staus of what is already running on the bus  )
I have d use clr halt on one HS and SS  h/w but in sequence ie clr halt , then rd from ep and did not face any issue so far (mostly window plus few  try on mac and linux). 

Even if it sound like it could be an usb controler os diver issue it is very unlikely ,  what is your h/w ?

Then how is the "clr halt translation" done at libusb level on linux  i rmenber it os dependent and properly done on mac (i guess window as well) but i have no clue on linux ? 
i imagine that crafting a "standard control request clr halt "  and sending it from user space would give different result than going true a "low level driver clr halt" 
I have use this hack to run do missing libusb "set halt" (opposite of  clr halt ie force stall on ep ) and it worked for a specific test case but it's not advisable to do so.

You may get more help or hint by  capturing  the low level kernel usb trace + debug and sending it  to the kernel usb list   or at first to libusb mail list .

dickens, 2018-03-25:
Data toggle is not controlled by libusb. If you suspect incorrect behavior in the host controller driver, please direct this to the Linux kernel USB mailing list.

closed 2018-03-25 by dickens