libusb/libusb #400: 1.0.22.rc3 Assync bulk RX never timer-out Win7-64

diabolo38, 2018-03-01:
I'm mixing Asynchronous bulk RX with Synchronous bulk Tx  basically 
- I prepare to  read device answer data by submiting an assynchronous Bulk RX from ep -1 IN  with time out 1sec 
- then i do a synch Bulk out to Ep-1 Out to send command to device it  succeed almost imediatly 
- Then i wait on  Call back  to fire with  completion or  time out with libusb_handle_events_completed
passing completed a ptr to volatile variable set by RX CB 
but this never return on 1.0.22rc3 , never Asynch xfer complete when device do not answer on time (i maintain device f/w under debugger breakpoint in purpose )

this do not show on 1.0.21 and earlier 

Process call stack when  "stuck on time-out" 

ntdll!ZwWaitForMultipleObjects() at 0x777d9e5a	
KERNELBASE!GetCurrentProcess() at 0x7fefd491430	
WaitForMultipleObjects() at 0x775606c0	
usbi_poll() at poll_windows.c:186 0x655537f5	
handle_events() at io.c:2,151 0x6555198d	
handle_events() at io.c:2,085 0x655528cb	
libusb_handle_events_timeout_completed() at io.c:2,348 0x655528cb	
libusb_handle_events_completed() at io.c:2,447 0x65552992	
<...more frames...>	

Below libusb log 
first part show a successful exchange with fast device answer

[ 2.349234] [00001854] libusb: debug [libusb_submit_transfer] transfer 00000000009799F8
[ 2.353235] [00001854] libusb: debug [usbi_add_pollfd] add fd 2 events 1
[ 2.357235] [00001854] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 81 with interface 1
[ 2.360236] [00001854] libusb: debug [winusbx_submit_bulk_transfer] reading 1024 bytes
[ 2.364236] [00001854] libusb: debug [libusb_alloc_transfer] transfer 0000000000979B18
[ 2.368236] [00001854] libusb: debug [libusb_submit_transfer] transfer 0000000000979B18
[ 2.371237] [00001854] libusb: debug [usbi_add_pollfd] add fd 3 events 4
[ 2.375237] [00001854] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 01 with interface 1
[ 2.379237] [00001854] libusb: debug [winusbx_submit_bulk_transfer] writing 13 bytes
[ 2.383238] [00001854] libusb: debug [libusb_get_next_timeout] next timeout in 0.970489s
[ 2.387238] [00001854] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 2.391239] [00001854] libusb: debug [handle_events] poll fds modified, reallocating
[ 2.394239] [00001854] libusb: debug [handle_events] poll() 3 fds with timeout in 971ms
[ 2.398239] [00001854] libusb: debug [handle_events] poll() returned 2
[ 2.401240] [00001854] libusb: debug [windows_handle_events] checking fd 2 with revents = 0001
[ 2.405240] [00001854] libusb: debug [usbi_remove_pollfd] remove fd 2
[ 2.408240] [00001854] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 0, size 6
[ 2.412241] [00001854] libusb: debug [usbi_handle_transfer_completion] transfer 00000000009799F8 has callback 0000000065542968
[ 2.416241] [00001854] libusb: debug [windows_handle_events] checking fd 3 with revents = 0004
[ 2.419241] [00001854] libusb: debug [usbi_remove_pollfd] remove fd 3
[ 2.422242] [00001854] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 0, size 13
[ 2.426242] [00001854] libusb: debug [usbi_handle_transfer_completion] transfer 0000000000979B18 has callback 00000000655530A0
[ 2.429242] [00001854] libusb: debug [sync_transfer_cb] actual_length=13
[ 2.433243] [00001854] libusb: debug [libusb_free_transfer] transfer 0000000000979B18

Log second part  with no answer from device expecting  time out 

[ 2.440244] [00001854] libusb: debug [libusb_submit_transfer] transfer 00000000009799F8
[ 2.443244] [00001854] libusb: debug [usbi_add_pollfd] add fd 2 events 1
[ 2.447244] [00001854] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 81 with interface 1
[ 2.450245] [00001854] libusb: debug [winusbx_submit_bulk_transfer] reading 1024 bytes
[ 2.454245] [00001854] libusb: debug [libusb_alloc_transfer] transfer 0000000000979B18
[ 2.457245] [00001854] libusb: debug [libusb_submit_transfer] transfer 0000000000979B18
[ 2.463246] [00001854] libusb: debug [usbi_add_pollfd] add fd 3 events 4
[ 2.467246] [00001854] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 01 with interface 1
[ 2.472247] [00001854] libusb: debug [winusbx_submit_bulk_transfer] writing 16 bytes
[ 2.476247] [00001854] libusb: debug [libusb_get_next_timeout] next timeout in 0.967069s
[ 2.481248] [00001854] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 2.486248] [00001854] libusb: debug [handle_events] poll fds modified, reallocating
[ 2.491249] [00001854] libusb: debug [handle_events] poll() 3 fds with timeout in 968ms
[ 2.496249] [00001854] libusb: debug [handle_events] poll() returned 1
[ 2.500250] [00001854] libusb: debug [windows_handle_events] checking fd 2 with revents = 0000
[ 2.505250] [00001854] libusb: debug [windows_handle_events] checking fd 3 with revents = 0004
[ 2.510251] [00001854] libusb: debug [usbi_remove_pollfd] remove fd 3
[ 2.514251] [00001854] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 0, size 16
[ 2.519251] [00001854] libusb: debug [usbi_handle_transfer_completion] transfer 0000000000979B18 has callback 00000000655530A0
[ 2.523252] [00001854] libusb: debug [sync_transfer_cb] actual_length=16
[ 2.527252] [00001854] libusb: debug [libusb_free_transfer] transfer 0000000000979B18
[ 2.531253] [00001854] libusb: debug [libusb_get_next_timeout] next timeout in 0.911469s
[ 2.536253] [00001854] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 2.540254] [00001854] libusb: debug [handle_events] poll fds modified, reallocating
[ 2.545254] [00001854] libusb: debug [handle_events] poll() 2 fds with timeout in 912ms
[ 3.461346] [00001854] libusb: debug [handle_events] poll() returned 0
[ 3.469346] [00001854] libusb: debug [libusb_cancel_transfer] transfer 00000000009799F8
[ 3.474347] [00001854] libusb: debug [winusbx_abort_transfers] will use interface 1
[ 3.480348] [00001854] libusb: error [winusbx_abort_transfers] CancelIoEx failed: [6] The handle is invalid.
[ 3.485348] [00001854] libusb: debug [libusb_cancel_transfer] cancel transfer failed error -4
[ 3.490349] [00001854] libusb: warning [handle_timeout] async cancel failed -4 errno=0
[ 3.496349] [00001854] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 3.501350] [00001854] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.506350] [00001854] libusb: debug [handle_events] poll() 2 fds with timeout in 60000ms
[63.516351] [00001854] libusb: debug [handle_events] poll() returned 0
[63.525351] [00001854] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[63.532352] [00001854] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[63.538353] [00001854] libusb: debug [handle_events] poll() 2 fds with timeout in 60000ms
[123.548353] [00001854] libusb: debug [handle_events] poll() returned 0
[126.006599] [00001854] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[126.013600] [00001854] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[126.020600] [00001854] libusb: debug [handle_events] poll() 2 fds with timeout in 60000ms
[186.028601] [00001854] libusb: debug [handle_events] poll() returned 0
[186.032601] [00001854] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[186.037601] [00001854] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[186.041602] [00001854] libusb: debug [handle_events] poll() 2 fds with timeout in 60000ms
[246.054603] [00001854] libusb: debug [handle_events] poll() returned 0
etc ... it never end

dickens, 2018-03-01:
What driver are you using for this interface? WinUSB? libusbK?

diabolo38, 2018-03-01:
it is libusbk

dickens, 2018-03-01:
I suspect that libusbk hasn't properly implemented cancellation through the native Windows interface. I will have a patch for you to try shortly.

dickens, 2018-03-01:
Could you see if this patch solves the issue? Thanks!

[winusbx_abort.zip](https://github.com/libusb/libusb/files/1772286/winusbx_abort.zip)

diabolo38, 2018-03-01:
seam to fix it  here is the log  maybe you can confirm i appluyed teh patch ok 
had to do it quick and need to leave office now :) 

[ 3.647364] [00001fd0] libusb: debug [libusb_submit_transfer] transfer 0000000000A1C888
[ 3.650365] [00001fd0] libusb: debug [usbi_add_pollfd] add fd 2 events 1
[ 3.654365] [00001fd0] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 81 with interface 1
[ 3.658365] [00001fd0] libusb: debug [winusbx_submit_bulk_transfer] reading 1024 bytes
[ 3.661366] [00001fd0] libusb: debug [libusb_alloc_transfer] transfer 0000000000A65D58
[ 3.665366] [00001fd0] libusb: debug [libusb_submit_transfer] transfer 0000000000A65D58
[ 3.668366] [00001fd0] libusb: debug [usbi_add_pollfd] add fd 3 events 4
[ 3.672367] [00001fd0] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 01 with interface 1
[ 3.676367] [00001fd0] libusb: debug [winusbx_submit_bulk_transfer] writing 16 bytes
[ 3.680368] [00001fd0] libusb: debug [libusb_get_next_timeout] next timeout in 0.970490s
[ 3.683368] [00001fd0] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.687368] [00001fd0] libusb: debug [handle_events] poll fds modified, reallocating
[ 3.690369] [00001fd0] libusb: debug [handle_events] poll() 3 fds with timeout in 971ms
[ 3.694369] [00001fd0] libusb: debug [handle_events] poll() returned 1
[ 3.697369] [00001fd0] libusb: debug [windows_handle_events] checking fd 2 with revents = 0000
[ 3.701370] [00001fd0] libusb: debug [windows_handle_events] checking fd 3 with revents = 0004
[ 3.704370] [00001fd0] libusb: debug [usbi_remove_pollfd] remove fd 3
[ 3.707370] [00001fd0] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 0, size 16
[ 3.711371] [00001fd0] libusb: debug [usbi_handle_transfer_completion] transfer 0000000000A65D58 has callback 0000000065550420
[ 3.714371] [00001fd0] libusb: debug [sync_transfer_cb] actual_length=16
[ 3.718371] [00001fd0] libusb: debug [libusb_free_transfer] transfer 0000000000A65D58
[ 3.721372] [00001fd0] libusb: debug [libusb_get_next_timeout] next timeout in 0.929004s
[ 3.726372] [00001fd0] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.730373] [00001fd0] libusb: debug [handle_events] poll fds modified, reallocating
[ 3.733373] [00001fd0] libusb: debug [handle_events] poll() 2 fds with timeout in 930ms
[ 4.667466] [00001fd0] libusb: debug [handle_events] poll() returned 0
[ 4.672467] [00001fd0] libusb: debug [libusb_cancel_transfer] transfer 0000000000A1C888
[ 4.680468] [00001fd0] libusb: debug [winusbx_abort_transfers] will use interface 1
[ 4.688468] [00001fd0] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 4.695469] [00001fd0] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 4.701470] [00001fd0] libusb: debug [handle_events] poll() 2 fds with timeout in 60000ms
[ 4.707470] [00001fd0] libusb: debug [handle_events] poll() returned 1
[ 4.715471] [00001fd0] libusb: debug [windows_handle_events] checking fd 2 with revents = 0001
[ 4.720472] [00001fd0] libusb: debug [usbi_remove_pollfd] remove fd 2
[ 4.727472] [00001fd0] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 995, size 0
[ 4.733473] [00001fd0] libusb: debug [windows_transfer_callback] detected operation aborted
[ 4.738473] [00001fd0] libusb: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
[ 4.743474] [00001fd0] libusb: debug [usbi_handle_transfer_completion] transfer 0000000000A1C888 has callback 0000000065542260
[E]  ASPIC_PlatformWrite 1167 :  bulkcom error -12
### FAIL ###  in run_gpio_testseq line 2819     set reset bit dw==0

commit 81ce8be5 (2018-03-12): Windows: WinUSB: Do not assume that driver implements CancelIoEx()

Cancelling an individual OVERLAPPED I/O operation is something that a
driver can optionally implement. It appears that the native WinUSB
driver supports this but libusbK does not. Therefore we will attempt
to use it and fall back to the AbortPipe() call if an error occurs.

Closes #400
closed 2018-03-12 by dickens