libusb/libusb #403: win7-64 enumeration listdev  bad device/duplicate

diabolo38, 2018-03-05:
on 1.0.22rc3 (also llikely rc2) 
my device appear multiple time in dev  list , only one is valid 
the second entry port/path is wrong and trying to open it will fail

i use win7-64 , Mingw64-64 bits 
usb bakcend is libusbk
device is composite but assigned device wise (ie vid/pid  inf file not per interface vid/pid/mixxx )

the issue is not present on 1.0.21
libusb example listdevs show the issue s

$ listdevs.exe
8086:1d2d (bus 4, device 0)
8086:1d26 (bus 3, device 0)
0409:005a (bus 3, device 2) path: 1.6
104c:8241 (bus 1, device 0)
0483:3748 (bus 3, device 6) path: 1.6.2
0553:0608 (bus 4, device 2) path: 1.3
0553:0608 (bus 1, device 2) path: 1 
413c:1003 (bus 3, device 3) path: 1.6.1
413c:2010 (bus 3, device 4) path: 1.6.1.1
413c:3016 (bus 3, device 5) path: 1.6.1.3
8087:0024 (bus 4, device 1) path: 1
8087:0024 (bus 3, device 1) path: 1
1033:0194 (bus 2, device 0)

0553:0608 (bus 4, device 2) path: 1.3 => is the good device 
0553:0608 (bus 1, device 2) path: 1=> is the "bad" device 

with Window/usbview debug utility the device show only once  Root-hub port 1 / Port 3 (path=1.3)

full log of listdev  below 
$ listdevs.exe
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [000004f8] libusb: debug [libusb_init] created default context
[ 0.001000] [000004f8] libusb: debug [libusb_init] libusb v1.0.22.11302-rc3
[ 0.002000] [000004f8] libusb: debug [windows_init_dlls] Will use CancelIoEx for I/O cancellation
[ 0.003000] [000004f8] libusb: debug [get_windows_version] Windows 7 SP1 64-bit
[ 0.004000] [000004f8] libusb: debug [windows_init_clock] hires timer available (Frequency: 2338154 Hz)
[ 0.005000] [000004f8] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.007000] [000004f8] libusb: debug [htab_create] using 1021 entries hash table
[ 0.011001] [000004f8] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.011001] [000004f8] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.013001] [000004f8] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.013001] [000004f8] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.014001] [000004f8] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.016001] [000004f8] libusb: info [windows_init] UsbDk backend is not available
[ 0.016001] [000004f8] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.017001] [000004f8] libusb: debug [libusb_get_device_list]
[ 0.120012] [000004f8] libusb: debug [winusb_get_device_list] allocating new device for session [9A0]
[ 0.121012] [000004f8] libusb: debug [winusb_get_device_list] found new PnP enumerator string 'NUSB3'
[ 0.121012] [000004f8] libusb: debug [winusb_get_device_list] allocating new device for session [25F8]
[ 0.122012] [000004f8] libusb: debug [winusb_get_device_list] allocating new device for session [26B8]
[ 0.122012] [000004f8] libusb: debug [winusb_get_device_list] allocating new device for session [26F4]
[ 0.123012] [000004f8] libusb: debug [winusb_get_device_list] allocating new device for session [2730]
[ 0.124012] [000004f8] libusb: debug [winusb_get_device_list] allocating new device for session [2780]
[ 0.125012] [000004f8] libusb: debug [winusb_get_device_list] allocating new device for session [27CC]
[ 0.126012] [000004f8] libusb: debug [winusb_get_device_list] allocating new device for session [281C]
[ 0.127012] [000004f8] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.128012] [000004f8] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.129012] [000004f8] libusb: debug [winusb_get_device_list] allocating new device for session [8A4]
[ 0.131013] [000004f8] libusb: debug [get_api_type] driver(s): libusbK
[ 0.131013] [000004f8] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.132013] [000004f8] libusb: debug [winusb_get_device_list] allocating new device for session [1A88]
[ 0.133013] [000004f8] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.134013] [000004f8] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.135013] [000004f8] libusb: debug [winusb_get_device_list] allocating new device for session [2080]
[ 0.136013] [000004f8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.137013] [000004f8] libusb: debug [get_api_type] matched driver name against Composite API
[... 133 lines cut ...]
0553:0608 (bus 1, device 1) path: 1 <= bad ???
413c:1003 (bus 3, device 3) path: 1.6.1
413c:2010 (bus 3, device 4) path: 1.6.1.1
413c:3016 (bus 3, device 5) path: 1.6.1.3
8087:0024 (bus 4, device 1) path: 1
8087:0024 (bus 3, device 1) path: 1
1033:0194 (bus 2, device 0)

0553:0608 (bus 1, device 1) path: 1 => loo, to be teh wron f device

below 2nd  hub  on on-board usb controler 
$ listdevs.exe
8086:1d2d (bus 4, device 0)
8086:1d26 (bus 3, device 0)
0409:005a (bus 3, device 2) path: 1.6
104c:8241 (bus 1, device 0)
0483:3748 (bus 3, device 6) path: 1.6.2
0553:0608 (bus 3, device 7) path: 1.5
0553:0608 (bus 1, device 7) path: 1
413c:1003 (bus 3, device 3) path: 1.6.1
413c:2010 (bus 3, device 4) path: 1.6.1.1
413c:3016 (bus 3, device 5) path: 1.6.1.3
8087:0024 (bus 4, device 1) path: 1
8087:0024 (bus 3, device 1) path: 1
1033:0194 (bus 2, device 0)

diabolo38, 2018-03-05:
actualy show with almost any device so likley not backend related
tried an fx2 non composite device,  same 
an usb3 non composite device same 
of the shelf phones (Galaxy J5, 1+ one ) same issue 

 $ listdevs.exe
8086:1d2d (bus 4, device 0)
8086:1d26 (bus 3, device 0)
0409:005a (bus 3, device 2) path: 1.6
104c:8241 (bus 1, device 0)
04e8:6860 (bus 4, device 2) path: 1.3
04e8:6860 (bus 1, device 2) path: 1
413c:1003 (bus 3, device 3) path: 1.6.1
413c:2010 (bus 3, device 4) path: 1.6.1.1
413c:3016 (bus 3, device 5) path: 1.6.1.3
8087:0024 (bus 4, device 1) path: 1
8087:0024 (bus 3, device 1) path: 1
1033:0194 (bus 2, device 0)

04e8:6860 (bus 4, device 2) path: 1.3 => Samsung Galaxy J5 phone 

surprisingly this doesn't show on 32 bit built :( 
first test were with gcc-64  6.3.0 , i tried again with 7.2.0 and 4.9.1 same ,

dickens, 2018-03-12:
@diabolo38 I suspect there is an issue with the third-party drivers provided by NEC. I don't think the device is being seen twice but rather the values on the stack from the previous enumeration were being reused. Can you run this again with the latest master?

diabolo38, 2018-03-12:
The device was on Intel controller not the NEC nor TI one. Also this is not parent on 1.0.21. Anyway I ll try on latest master and let you know.

mcuee, 2018-03-13:
You are using libusbK driver. Is it possible you switch to WinUSB driver to see if there is any difference or not? If not, delete libusbK.dll from the system and then try again to see if that makes any difference or not.

diabolo38, 2018-03-13:
@mcuee It  was the same even with no devcie no back end driver at all (J5 android phone show same) 

@dickens confirmed it's fixed  many thanks for that I let you close the issue 

listdev output 
$ listdevs.exe
8086:1d2d (bus 4, device 0)
8086:1d26 (bus 3, device 0)
0409:005a (bus 3, device 3) path: 1.6
104c:8241 (bus 1, device 0)
0483:3748 (bus 3, device 4) path: 1.6.2
0553:0608 (bus 3, device 1) path: 1.5
413c:1003 (bus 3, device 5) path: 1.6.1
413c:2010 (bus 3, device 2) path: 1.6.1.1
413c:3016 (bus 3, device 6) path: 1.6.1.3
8087:0024 (bus 4, device 2) path: 1
8087:0024 (bus 3, device 8) path: 1
1033:0194 (bus 2, device 0)

full log 
$ listdevs.exe
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [0000184c] libusb: debug [libusb_init] created default context
[ 0.007000] [0000184c] libusb: debug [libusb_init] libusb v1.0.22.11307-rc3
[ 0.010001] [0000184c] libusb: debug [windows_init_dlls] Will use CancelIoEx for I/O cancellation
[ 0.012001] [0000184c] libusb: debug [get_windows_version] Windows 7 SP1 64-bit
[ 0.014001] [0000184c] libusb: debug [windows_init_clock] hires timer available (Frequency: 2338134 Hz)
[ 0.016001] [0000184c] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.019001] [0000184c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.023002] [0000184c] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.026002] [0000184c] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.029002] [0000184c] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.031003] [0000184c] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.034003] [0000184c] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.038003] [0000184c] libusb: info [windows_init] UsbDk backend is not available
[ 0.041004] [0000184c] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.043004] [0000184c] libusb: debug [libusb_get_device_list]
[ 0.127012] [0000184c] libusb: debug [winusb_get_device_list] allocating new device for session [9A0]
[ 0.130013] [0000184c] libusb: debug [winusb_get_device_list] found new PnP enumerator string 'NUSB3'
[ 0.132013] [0000184c] libusb: debug [winusb_get_device_list] allocating new device for session [25F8]
[ 0.134013] [0000184c] libusb: debug [winusb_get_device_list] allocating new device for session [26B8]
[ 0.136013] [0000184c] libusb: debug [winusb_get_device_list] allocating new device for session [26F4]
[ 0.138013] [0000184c] libusb: debug [winusb_get_device_list] allocating new device for session [2730]
[ 0.140014] [0000184c] libusb: debug [winusb_get_device_list] allocating new device for session [2780]
[ 0.142014] [0000184c] libusb: debug [winusb_get_device_list] allocating new device for session [27CC]
[ 0.144014] [0000184c] libusb: debug [winusb_get_device_list] allocating new device for session [281C]
[ 0.146014] [0000184c] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.149014] [0000184c] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.151015] [0000184c] libusb: debug [winusb_get_device_list] allocating new device for session [8A4]
[ 0.154015] [0000184c] libusb: debug [get_api_type] driver(s): libusbK
[ 0.156015] [0000184c] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.159015] [0000184c] libusb: debug [winusb_get_device_list] allocating new device for session [1A88]
[ 0.161016] [0000184c] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.164016] [0000184c] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.166016] [0000184c] libusb: debug [winusb_get_device_list] allocating new device for session [2080]
[ 0.168016] [0000184c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.170017] [0000184c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.172017] [0000184c] libusb: debug [winusb_get_device_list] allocating new device for session [20CC]
[ 0.174017] [0000184c] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.176017] [0000184c] libusb: debug [get_api_type] matched driver name against HID API
[ 0.178017] [0000184c] libusb: debug [winusb_get_device_list] allocating new device for session [2118]
[ 0.180018] [0000184c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_104C&DEV_8241&SUBSYS_158B103C&REV_02\4&2FA50F60&0&00E6' bus numbe
r 1
[ 0.184018] [0000184c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_1033&DEV_0194&SUBSYS_00000000&REV_03\4&2E92337F&0&00E5' bus numbe
r 2
[ 0.188018] [0000184c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_1D26&SUBSYS_158B103C&REV_05\3&11583659&0&E8' bus number
3
[ 0.191019] [0000184c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_1D2D&SUBSYS_158B103C&REV_05\3&11583659&0&D0' bus number
4
[... 79 lines cut ...]
413c:2010 (bus 3, device 2) path: 1.6.1.1
[ 0.383038] [0000184c] libusb: debug [libusb_get_device_descriptor]
413c:3016 (bus 3, device 6) path: 1.6.1.3
[ 0.387038] [0000184c] libusb: debug [libusb_get_device_descriptor]
8087:0024 (bus 4, device 2) path: 1
[ 0.391039] [0000184c] libusb: debug [libusb_get_device_descriptor]
8087:0024 (bus 3, device 8) path: 1
[ 0.394039] [0000184c] libusb: debug [libusb_get_device_descriptor]
1033:0194 (bus 2, device 0)
[ 0.398039] [0000184c] libusb: debug [libusb_unref_device] destroy device 1.0
[... cut ...]
[ 0.413041] [0000184c] libusb: debug [libusb_unref_device] destroy device 3.8
[ 0.415041] [0000184c] libusb: debug [libusb_unref_device] destroy device 3.0
[ 0.417041] [0000184c] libusb: debug [libusb_unref_device] destroy device 2.0
[ 0.418041] [0000184c] libusb: debug [libusb_exit]
[ 0.420042] [0000184c] libusb: debug [libusb_exit] destroying default context
[ 0.422042] [0000184c] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.424042] [00001ccc] libusb: debug [windows_clock_gettime_threaded] timer thread quitting

dickens, 2018-03-14:
@diabolo38 thanks for trying it out and reporting back.

I'm curious about the following line from the log:

`[ 0.244024] [0000184c] libusb: error [init_device] device 'USB\VID_21A9&PID_1004\6&17C4364E&0&4' is no longer connected!`

Indeed this device is not being listed. Can you post the debug log of this same test using libusb v1.0.21? Also, can you let me know which hub/port and HCD this device is connected to as reported by usbview?

Thanks!

diabolo38, 2018-03-15:
It  must  be coming from a device on ti usb3 controllers.  Only one of its port successfully  report device the other tend to give this type of error message. The working port is random most of the time 0 work  but some time it is 1
This is very old issue I reported on the mail list long time ago and has even some older threads . I give up trying to use libusb o  this controllers  under windows as it is quite rare nowadays.

dickens, 2018-03-15:
Ok, I'll close this issue then.

closed 2018-03-15 by dickens