libusb/libusb #405: Error while compiling libusb_get_config_descriptor

arindamsetubiswas, 2018-03-11:
Hi everyone,

I am doing this project in which I am using libusb. every function in the class seems to work but the function libusb_get_config_descriptor is giving me an error and it's unusual.

here is my code..... please help.

#include <iostream>
#include <libusb.h>
#include <cassert>
#include <cstdio>
using namespace std;

using namespace std;

void print_devices(libusb_device *dev)
{
	libusb_device_descriptor desc;
	libusb_interface *inter;
	libusb_config_descriptor *config;
	libusb_interface_descriptor *interdesc;
	libusb_endpoint_descriptor *endpointdesc;

	int ret;
	int i,j,k;

	ret=libusb_get_device_descriptor(dev,&desc);

	if(ret<0)
	{
	 	fprintf(stderr,"error in getting device descriptor\n");

	}

	printf("Number of pconfigs is %d\n",desc.bNumConfigurations);
	printf("Device class: %d\n",desc.idVendor);
	printf("Product Id: %d\n",desc.idProduct);

	libusb_get_config_descriptor(dev,0,&config);///////////error
	cout<<"interfaces : "<<(int)config->bNumInterfaces<<endl;

	printf("\n\n");

	libusb_free_config_descriptor(config);

}

int main(int argc, char *argv[])
{
	libusb_device **devs;
	libusb_context *context=NULL;

	size_t list;
	size_t i;
	int ret;

	ret=libusb_init(&context);

	if(ret<0)
	{
		perror("libusb_init");
		exit(1);
	}

	list=libusb_get_device_list(context, &devs);

	if(list<0){
		fprintf(stderr,"Error in getting device list\n");
		libusb_free_device_list(devs,1);
		libusb_exit(context);
		exit(1);
	}

	printf("There are %d devices found\n",list);

	for(i=0;i<list; i++)
	{
		//print devices specs
		print_devices(devs[i]);

	}

	libusb_free_device_list(devs,1);
	libusb_exit(context);

	return 0;
}

LudovicRousseau, 2018-03-11:
Do we have to guess the error code you get?
Please read https://github.com/libusb/libusb/wiki/FAQ#getting-help

dickens, 2018-03-25:
No response from the OP, so closing this. This can be re-opened with a helpful log file.

closed 2018-03-25 by dickens