libusb/libusb #410: Build failure on OS X 10.7: use of undeclared identifier 'kUSBDeviceSpeedSuper'

ryandesign, 2018-03-25:
libusb 1.0.22 fails to build on OS X 10.7 with:

```
os/darwin_usb.c:1043:10: error: use of undeclared identifier 'kUSBDeviceSpeedSuper'; did you mean 'kUSBDeviceSpeedFull'?
    case kUSBDeviceSpeedSuper: dev->speed = LIBUSB_SPEED_SUPER; break;
         ^~~~~~~~~~~~~~~~~~~~
         kUSBDeviceSpeedFull
/System/Library/Frameworks/IOKit.framework/Headers/usb/USB.h:899:9: note: 'kUSBDeviceSpeedFull' declared here
        kUSBDeviceSpeedFull             = 1,
        ^
```

Here is a [full build log](https://build.macports.org/builders/ports-10.7_x86_64_legacy-builder/builds/61294/steps/install-port/logs/stdio) from the MacPorts project's automated build system.

This is a new problem for 1.0.22; 1.0.21 didn't have this problem.

It's caused by e4851c38946560ac0fe3eeea848b37c264b4a9c8 which changes the restriction on when `kUSBDeviceSpeedSuper` is used to the condition `MAC_OS_X_VERSION_MAX_ALLOWED >= 1070`. [Apple documentation](https://developer.apple.com/documentation/iokit/1425357-usbdevicespeed/kusbdevicespeedsuper) says `kUSBDeviceSpeedSuper` was introduced in OS X 10.7, but the documentation is apparently incorrect. According to [a SourceForge mailing list post](https://sourceforge.net/p/libusb/mailman/message/30467617/), it was actually introduced in 10.8. libusb 1.0.22 does build successfully in MacPorts on OS X 10.8. So I would change the condition to `MAC_OS_X_VERSION_MAX_ALLOWED >= 1080`.

mcuee, 2018-03-25:
That Sourceforge post is from me and at that time I can not build libusbus under Mac OS X 10.7. So your experience with libusb-1.0.22 is similar. My Mac Mini 2011 has been upgraded to the latest version so I no longer be able to confirm this. But I think your change seems to be good.

seanm, 2019-04-12:
I believe this is fixed as of at least 1.0.23-rc1.

hjelmn, 2019-04-12:
Ok, let me know. If it doesn't work keep in mind 10.7 is considered obsolete by the vendor and is no longer supported. If it is still broken and someone else (not me) opens a PR to fix it I will certainly merge it.

seanm, 2020-03-20:
This is fixed in master.

mcuee, 2020-07-29:
Close this one.

ryandesign, 2021-06-12:
> This is fixed in master.

I do not see such a fix in master. MacPorts continues to apply [the patch](https://github.com/macports/macports-ports/blob/af475b94db3fbb41b40d56e2bf93b595bcc0ad76/devel/libusb/files/patch-10.7-nospeedsuper.diff). Please reopen until the patch is applied to master. (Or do you think a fix was made somewhere else? In what commit?)

mcuee, 2021-06-12:
@ryandesign Please help to create a pull request based on the MacPorts patch so that we have a record for the patch. Thanks.

seanm, 2021-06-12:
@mcuee it's conditionalized here:

https://github.com/libusb/libusb/blob/5360a17ce7a596b0fb976a1be6a2b51753e87824/libusb/os/darwin_usb.c#L1146

seanm, 2021-06-12:
Oh, I see... you're saying it should be 10.8, not 10.7... in that case, yeah, I guess a PR is needed.

ryandesign, 2021-06-12:
Pretty frustrating.

commit 68289d0d (2021-07-09): Fix build failure on Mac OS X 10.7

Apple documentation says kUSBDeviceSpeedSuper is available in 10.7 and
later but in fact it is only defined in the 10.8 SDK and later.

Closes #410
Closes #929
closed 2021-07-09 by hjelmn