libusb/libusb #411: When release 1.0.22 will be available on PyPI?

vasily-v-ryabov, 2018-03-27:
Please post latest release to PyPI. We'd like to use final release through `pip` instead of beta. Thanks!

mcuee, 2018-03-27:
Wrong one. libusb has nothing to do with PyPi. Maybe you are talking about
pyusb or python-libusb1. Will close this.

> --
Xiaofan

closed 2018-03-27 by mcuee

vasily-v-ryabov, 2018-03-27:
Sorry, it seems there is different maintainer for Python bindings:
https://pypi.python.org/pypi/libusb

mcuee, 2018-03-28:
Interestingly there is new python binding for libusb. Just wondering what is the advantage compare to the more popular pyusb. There is also python-libusb1.

pyusb
https://pypi.python.org/pypi/pyusb

python-libusb1
https://pypi.python.org/pypi/libusb1

vasily-v-ryabov, 2018-05-10:
Well, we decided to use C version as a Git submodule for now. There were no deep comparison of these bindings. Thanks for the links!